mne-tools/mne-python #1892: Importing bdf files greater >2gb

brauner, 2015-03-23:
Hello,

is there any way to import `*.bdf` files which are greater than `2gb`. On the data conversion page on the homepage it is stated:

>Warning

>The data samples in a BDF file are represented in a 3-byte (24-bit) format. Since 3-byte raw data buffers are not presently supported in the fif format these data will be changed to 4-byte integers in the conversion. Since the maximum size of a fif file is 2 GBytes, the maximum size of a BDF file to be converted is approximately 1.5 GBytes

Is there a way that `mne-python` can import it an split it automatically into separate `*.fif` files that all get loaded together whenever one of them gets imported?

teonlamont, 2015-03-23:
hey @brauner I think that is what is currently implemented in mne-python. Could you try reading in a .bdf and saving the raw file? Our saving function should take care of this automatically.

brauner, 2015-03-23:
Cheers, @teonlamont I'll try it now and report back.

teonlamont, 2015-03-23:
@brauner I just noticed that it was a warning for MNE command line tools. That warning as well as the uneven sampling rate are no longer a problem with mne-python.

brauner, 2015-03-23:
@teonlamont, ok. But I got a strange problem. I cannot find the function `mne_edf2fiff()`. I'm on `Ubuntu` with python-mne installed from the package manager. I updated it with `pip` and even changed to the `git` version with `pip`. Still no luck. Any hints?

teonlamont, 2015-03-23:
in mne-python, it's `mne.io.read_raw_edf()`

brauner, 2015-03-23:
d1a1_raw = mne.io.read_raw_edf('/home/ubuntu/E/d1a1.bdf')
    Extracting edf Parameters from /home/ubuntu/E/d1a1.bdf...
    Setting channel info structure...
    Creating Raw.info structure...
    Ready.

`d1a1.bdf` is `2.9gb` and is processed in under a second (`8gb RAM` on my system)... Expected speed or suspicious?

teonlamont, 2015-03-23:
by default, we set `preload=False`. This loads all of the data header information but not the data themselves. if you would like to load the data, there are two options.

```
raw = mne.io.read_raw_edf(..., preload=False)
raw.preload_data()
```

or 
```
raw = mne.io.read_raw_edf(..., preload=True)
```

brauner, 2015-03-23:
Thanks, @teonlamont. How can I save

    raw = mne.io.read_raw_edf(..., preload=False)

to a `*.fiff` file. I'm having a hard time finding this in the manual.

teonlamont, 2015-03-23:
no prob! to save to disk, you can use the method,
raw.save('filename-raw.fif'). this is typically the format that we use to
name files.

Unfortunately, the mne-manual doesn't have any of the python capabilities
listed.

-teon

teonlamont, 2015-03-23:
if there are no other problems, I am going to close this issue.

closed 2015-03-23 by teonlamont

brauner, 2015-03-23:
Cheers, @teonlamont but I cannot find a `raw.save('filename-raw.fif')` method. Is it called differently?

brauner, 2015-03-27:
Quick feedback, `mne-python` correctly reads in `*.bdf` files greater than `2gb`, splits them up into multiple `*.fif` files when saving, and reads all of them in together. Thanks again, @teonlamont.

teonlamont, 2015-03-27:
@brauner i'm glad to hear it! :) and no prob!