mne-tools/mne-python #1904: Q: saving GAT object

kingjr, 2015-03-26:
@dengemann @agramfort 

I have started looking at how we should save the GeneralizationAcrossTime objects.

I originally thought of using a simple pickle.dump it, but the disk usage really quickly climbs up (mainly because of the number of estimators). 

Would you have any tips to circumvent this issue? For instance, as by default we use clf=pipeline(scaler, svc), I thought of scaling the svc.coef_ and saving a pipeline(scaled_svc) so as to avoid keeping scaler.mean_ and scaler.std_ (dims = 2 x n_features x training_times x n_folds), but I'm not sure whether this is a useful and valid approach.

agramfort, 2015-03-26:
I am afraid pickle is the less ugly way. You can compress though. The other
solution will be too custom and fragile I suspect.

kingjr, 2015-03-27:
Ok thanks

closed 2015-03-27 by kingjr