mne-tools/mne-python #1906: bug IO with evoked.

agramfort, 2015-03-26:
use this file

https://dl.dropboxusercontent.com/u/2140486/toto-ave.pkl

and run

```
import cPickle as p
a = p.load(open('toto-ave.pkl', 'r'))
a.save('pouet-ave.fif')
mne.read_evokeds('pouet-ave.fif')
```

can somebody have look?

@Eric89GXL maybe?

Eric89GXL, 2015-03-27:
I can't even load the file in Python3. Perhaps it is a unicode issue -- was this created on latest master with the unicode fixes, or no?

agramfort, 2015-03-27:
I can load it on python 2 with current master

Eric89GXL, 2015-03-27:
Do you know if it was created before or after the unicode fixes?

agramfort, 2015-03-27:
no idea

agramfort, 2015-03-31:
ok it's again a problem of IO of sparse data.

If I do

```
import cPickle as p
a = p.load(open('toto-ave.pkl', 'r'))
del a.info['proc_history'][0]['max_info']['sss_ctc']['decoupler']
a.save('pouet-ave.fif')
import mne
mne.read_evokeds('pouet-ave.fif')
```

a.info['proc_history'][0]['max_info']['sss_ctc']['decoupler'] is a CSR sparse matrix.

@mluessi any chance you can have a look?

agramfort, 2015-03-31:
looks like the a bad file from MNE-C slips through our reading fix.

I have need to have access to the raw file obtained with mne_process_raw...

virvw, 2015-03-31:
that is strange, the file was NOT processed with mne_process_raw but I do use mne_browse_raw or xfit or ... to visualize

agramfort, 2015-03-31:
hum... how can I get access to the initial raw file?

virvw, 2015-03-31:
you can find one exemple of "run2" but the ave is drawn fom several such
runs

https://drive.google.com/folderview?id=0B7AsPwL-EHVlYWY0Nzc5OGQtYjY2MC00YThmLThjNWItMTBmNTJhN2JhN2E4&usp=sharing

agramfort, 2015-03-31:
@virvw should be fixed with

https://github.com/mne-tools/mne-python/commit/4529854f0518e2393e666e7fc5b0aef33edc8c38

reopen the issue if not

closed 2015-03-31 by agramfort

virvw, 2015-04-01:
@agramfort  I confirm it is fixed. Generated files readable back in python and mne gui suite.

agramfort, 2015-04-01:
awesome !