mne-tools/mne-python #1907: raw.filter docstring

teonlamont, 2015-03-27:
I am wondering if I am misunderstanding something or if the docstring is incorrect.

there's a section of the docstring that says:
```
    l_freq is not None, h_freq is None: low-pass filter
    l_freq is None, h_freq is not None: high-pass filter
```

but under the parameters:
```
l_freq : float | None
    Low cut-off frequency in Hz. If None the data are only low-passed.
h_freq : float | None
    High cut-off frequency in Hz. If None the data are only
    high-passed.
```

Those are conflicting statements. Which one is correct?

Eric89GXL, 2015-03-27:
Argh, I thought we sorted this out already. Seems to come up repeatedly... who chose these variable names anyway? :)

`h_freq` sets the highest frequency in the passband, therefore the lowpass frequency. `l_freq` sets the lowest frequency in the passpand, thus the highpass frequency. So the parameter docstring is correct, the additional docstring is incorrect.

teonlamont, 2015-03-27:
these variables are awful and give me unnecessary headaches.

I will update the docstring to fix the error.

-teon

teonlamont, 2015-03-27:
closing for #1908.

closed 2015-03-27 by teonlamont

agramfort, 2015-03-27:
>
> these variables are awful and give me unnecessary headaches
>

they make/made sense to me ... sorry... :-/

dengemann, 2015-03-27:
for me too ....