mne-tools/mne-python #1909: Converting channel types on the fly?

choldgraf, 2015-03-28:
I'm reading in an EDF file where the analog channels are labeled as EEG. I'm trying to figure out how to set these to misc, but haven't found an easy way to do this. I've learned that it's usually not a good idea to directly change fields in the "info" attribute, so I wonder if there's a good way to do this that I'm not thinking of?

dengemann, 2015-03-29:
@choldgraf take a look at `mne.rename_channels`. Maybe it can do the job for you.

choldgraf, 2015-03-29:
ah perfect. out of curiosity why can't you change the type to eeg?

dengemann, 2015-03-29:
> why can't you change the type to eeg?

Hum?

choldgraf, 2015-03-29:
from the docstring (I also tried it and it returned an error)

>    """Rename channels and optionally change the sensor type.
    Note: This only changes between the following sensor types: eeg, eog,
    emg, ecg, and misc. **It also cannot change to eeg.**
    Parameters
    ----------
    info : dict
        Measurement info.
    mapping : dict
        a dictionary mapping the old channel to a new channel name {'EEG061' :
        'EEG161'}. If changing the sensor type, make the new name a tuple with
        the name (str) and the new channel type (str)
        {'EEG061':('EOG061','eog')}.

dengemann, 2015-03-29:
@dgwakeman ??

agramfort, 2015-03-29:
this is crazy. It should not be the case.

alos eeg should be in human2fiff dict in function.

PR more than welcome !!!

choldgraf, 2015-03-29:
> @choldgraf <https://github.com/choldgraf> there's a parameter, misc in
> all the eeg readers when you first load the data.

Ah that's useful as well, thanks :)

dgwakeman, 2015-03-30:
Yeah, that is a typo. It is meant to say MEG. I'll fix it later this week. Unless someone else wants to tackle this easy PR.

agramfort, 2015-03-30:
@dgwakeman it's 5 mins of your time. Why does it have to wait a week???

dgwakeman, 2015-03-30:
I'm busy (and not paid to work on MNE). Also, I just realized it's not a typo. There is a good reason:
To define an EEG sensor you need to adjust dig and define a new sensor. That is more complex. It should IMO go in a new function e.g. mne.channels.define_sensor

dengemann, 2015-03-30:
>  mne.channels.define_sensor

sounds like a useful API. We could then leave it to the user to provide as much of the information as needed. If nothing additional is provided we would just use defaults and change the sensor type.

teonlamont, 2015-03-31:
wait, @dgwakeman why should dig be adjusted? dig is usually reserved for digitized head points and hpi points, not for sensors.

Eric89GXL, 2015-03-31:
I think `eeg` channels end up in our `info['dig']` when we do M/EEG recordings

dgwakeman, 2015-03-31:
@teonlamont: yes what @Eric89GXL said, plus you need to modify channels, which gets complex fast when you are really changing channel type. The channel type changes mentioned here are really more about name than type (the difference between eog, emg, ecg, and misc is not really important for anything except as a name: same units, attributes etc.). Hence, why I think changing the type should be a different API, this was just a weird exception due to hardware vendors. 

IMO, it is a mistake to incorporate SEEG in this function, as these should be done properly as sensor types, and should be added to dig etc. Perhaps, this discussion will spark some inspiration in people.

agramfort, 2015-03-31:
https://www.youtube.com/watch?v=BKorP55Aqvg

:) :)

choldgraf, 2015-03-31:
Do you think that you could implement the ability to rename channel types to "eeg" in C++, with pen and paper?

agramfort, 2015-03-31:
;)

teonlamont, 2015-04-06:
there's now pr #1932 to tackle this issue. currently it doesn't deal eeg channels in the `info['dig']`.

closed 2015-04-10 by teonlamont