mne-tools/mne-python #1913: Support for .hpts broken

haribharadwaj, 2015-03-30:
Hmm.. so since ```read_raw_edf``` started taking ```montage``` instance as input, support for .hpts is broken.  ```channels.read_montage``` doesn't resolve cardinal points and non-EEG channels correctly leading to not being able to use ```transform=True```.. Other things fail downstream too..

agramfort, 2015-03-31:
@teonlamont @christianbrodbeck @wmvanvliet ?

teonlamont, 2015-03-31:
ok, currently it was transforming all channels so we can resolve it within the function by applying picks. what are some of the downstream effects or is it that not resolving the transformation leads to a downstream effect?

haribharadwaj, 2015-03-31:
@teonlamont .. I just looked through the code and it seems like the identifiers it is looking for the cardinals is ```lpa```, ```rpa``` and ```nasion```. The spec for hpts in the manual says "For cardinal landmarks, 1 = left auricular point, 2 = nasion, and 3 = right auricular point." I can quickly edit to look for either the string or the digit...

The downstream issues seem to be because of this too. Sorry about the alarm.

haribharadwaj, 2015-03-31:
Closed with PR #1918

closed 2015-03-31 by haribharadwaj