mne-tools/mne-python #1917: ENH/FIX: dealing with huge effect sizes in linear regression

dengemann, 2015-03-31:
In some cases if you encounter t-values around 17 the scipy cdf estimator yields values of 1. This leads to infs in then -log10(p value) computation. I'm wondering whether we should deal with this internally somehow.

Eric89GXL, 2015-03-31:
perhaps you should regularize your estimates e.g. by using the "hat" option in the t-test calculation

dengemann, 2015-03-31:
> perhaps you should regularize your estimates e.g. by using the "hat" option in the t-test calculation

we could do that inside our regression function maybe ...
On the other hand I'm wondering what it tells you if your intercept estimate has such crazy t-values ... maybe some offset or baselining issue.

Eric89GXL, 2015-03-31:
Yeah, either the means are way off (maybe baselining or offset) or the standard deviation estimate is too small to be realistic (could be fixed by "hat")

PR #2713 (2015-12-17): MRG: Fix regression
Closes #2056.
Closes #1917.

Gets rid of warnings about divide by zero and underflows in linear regression by protecting against `stderr == 0`. It effectively forces all values in mlog10_p_val to be in the range `[-9.64e-17, 0]` (i.e., `[np.log10(1. - np.finfo(np.float64).eps), 0]`).

@dengemann I'd like to also take care of #1917 with this, but practically I'm not sure if we should make any changes to the code -- it might be doing the right thing already. I think if you get a p-value of 1., you should get a `0` in your `mlog10_p_val` and `np.inf` in your `t_val`, and deal  [... cut]
closed 2015-12-17 by dengemann