mne-tools/mne-python #1926: auto gen layouts

teonlamont, 2015-04-02:
I'm trying to use `plot_topo` with KIT data from our new system that has more sensors than the 157 one but it gives an error because there's no predefined layout. is there a way for one to be generated using the coordinates? I know that `plot_ica_components` renders the channel positions automatically, maybe something like that?

dengemann, 2015-04-02:
> I know that plot_ica_components renders the channel positions automatically, maybe something like that?

so spot the differences in the :)

I know that it can do so if the sensor positions from the info['chs'] are considered and projected onto 2d, we have code to do that.

teonlamont, 2015-04-02:
haha, ok, yeah, I was just making sure that there wasn't a function that didn't already do this. I feel like i'm a new function everyday ;) 

how should we handle the fix? maybe we should have a function that's responsible for flattening 3d surfaces that other functions can pull from instead of copying code?

dengemann, 2015-04-02:
@teonlamont we have all this. Please just look it up and make a PR.

teonlamont, 2015-04-02:
ok, that's cool with me, it was just that I couldn't tell whether this was
a viz function or a layout because they both do similar things for
different purposes. i wanted to make sure I was needlessly reproducing code.

i guess the most fitting would be make_eeg_layout but it seems like this
title is a bit restricting. i'll make a pr.

-teon

dengemann, 2015-04-02:
@teonlamont sorry I'm hacking on other stuff at this point. Once I finde a break I'll share a pointer with you. Just see how it's done in the ICA function.

agramfort, 2015-04-03:
See eg in fieldtrip if there is no existing .lay file for your system

I would work to have this .lay file

wmvanvliet, 2015-04-03:
the `make_eeg_layout` function does what you want, but for EEG channels. It may be desirable to generalise this function to work on any channel type.

wmvanvliet, 2015-04-03:
but constructing a proper .lay file is probably the better approach.

agramfort, 2015-04-03:
for MEG we have a few possible layouts but let's use template files.

agramfort, 2015-04-03:
+100

teonlamont, 2015-04-07:
I tired using one of the built-in templates that's in fieldtrip for yokogawa system (KIT) but I don't think these are designed for our system. The layout turned out very strange. what would be the disadvantages of using an automated layout from the sensor info similar to what we do for the eeg?

agramfort, 2015-04-07:
Generate once for you system and create the lay file. Maybe ask on fieldtrip ml if this file exists somewhere and if not offer to share yours with them

My 2c

wmvanvliet, 2015-04-07:
The disadvantage is that auto generated layouts are usually not optimal. I find it often beneficial to tweak the location of the plots a little so there is less whitespace between them.

teonlamont, 2015-04-19:
but there should be an option for an auto layout when the `.lay` or `.lout` are not available. Then the `find_layout` would then just default to `make_eeg_layout` when the optimal layout is not available.

agramfort, 2015-04-19:
isn't it the case already?

teonlamont, 2015-04-19:
not for MEG, it just returns none. It works like this for EEG.

agramfort, 2015-04-19:
for MEG we should have a layout for each system. There are not so many.

teonlamont, 2015-04-19:
for some reason, I can't find the proper layout for the new KIT in Abu Dhabi. The one in fieldtrip has layout constructed of 440 and our system only has 208. I tried taking a selection of the channels but it rendered a very unusable layout. maybe I need to dig more into how the plots are laid out in fieldtrip and try to port a proper one that way.

agramfort, 2015-04-19:
+1

generate one and we can improve it afterwards

teonlamont, 2015-04-19:
I just remembered why I originally made the pr. I can plot my topomaps with `evoked.plot_topomap` without a layout b/c there is a `_find_topomap_coords`. with `plot_topo`, it needs a layout. I will work on the layout and make a pr.

closed 2015-04-19 by teonlamont