mne-tools/mne-python #1927: Example plot_mne_inverse_coherence_epochs.py crashes python

breuderink, 2015-04-03:
Hi,

I am completely new to MNE; I want to use it to visualise source activations for a sensor-level vector autoregressive model. To start I tried to reproduce this picture:

```
$ python -V
Python 2.7.9
$ BR-MBA-2014:tmp borisreuderink$ pip freeze | grep mne
mne==0.8.6

curl -O http://martinos.org/mne/stable/_downloads/plot_mne_inverse_coherence_epochs.py
$ python plot_mne_inverse_coherence_epochs.py
[lots of text here]
```
[image]

Is this a bug in MNE? Or in one of it's dependencies?

Eric89GXL, 2015-04-03:
I have never seen it. mne-python is pure Python, so it is probably a problem with a dependency (joblib?). Can you do `import faulthandler; faulthandler.enable()` at the top of your script and try again? It might give us a stack trace.

breuderink, 2015-04-03:
This is what I get with faulthandler: 

```
Processing epoch : 2
    computing connectivity for 7498 connections
    using t=-0.200s..0.499s for estimation (106 points)
    computing connectivity for the bands:
     band 1: 8.5Hz..12.7Hz (4 points)
     band 2: 14.2Hz..29.7Hz (12 points)
    connectivity scores will be averaged for each band
    using FFT with a Hanning window to estimate spectra
    the following metrics will be computed: Coherence
    computing connectivity for epochs 1..2
Fatal Python error: Segmentation fault

Current thread 0x00007fff7eab5300 (most recent call first):
  File "/usr/local/lib/python2.7/site-packages/mne/source_estimate.py", line 786 in transform_data
  File "/usr/local/lib/python2.7/site-packages/mne/connectivity/spectral.py", line 333 in _epoch_spectral_connectivity
  File "/usr/local/lib/python2.7/site-packages/sklearn/externals/joblib/parallel.py", line 92 in __call__
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/pool.py", line 113 in worker
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 114 in run
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 258 in _bootstrap
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/forking.py", line 126 in __init__
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 130 in start
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/pool.py", line 223 in _repopulate_pool
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/pool.py", line 159 in __init__
  File "/usr/local/lib/python2.7/site-packages/sklearn/externals/joblib/pool.py", line 402 in __init__
  File "/usr/local/lib/python2.7/site-packages/sklearn/externals/joblib/pool.py", line 563 in __init__
  File "/usr/local/lib/python2.7/site-packages/sklearn/externals/joblib/parallel.py", line 613 in __call__
  File "/usr/local/lib/python2.7/site-packages/mne/connectivity/spectral.py", line 990 in spectral_connectivity
  File "/usr/local/lib/python2.7/site-packages/mne/utils.py", line 551 in verbose
  File "<string>", line 2 in spectral_connectivity
  File "plot_mne_inverse_coherence_epochs.py", line 100 in <module>
Fatal Python error: Segmentation fault

Current thread 0x00007fff7eab5300 (most recent call first):
  File "/usr/local/lib/python2.7/site-packages/mne/source_estimate.py", line 786 in transform_data
  File "/usr/local/lib/python2.7/site-packages/mne/connectivity/spectral.py", line 333 in _epoch_spectral_connectivity
  File "/usr/local/lib/python2.7/site-packages/sklearn/externals/joblib/parallel.py", line 92 in __call__
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/pool.py", line 113 in worker
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 114 in run
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 258 in _bootstrap
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/forking.py", line 126 in __init__
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 130 in start
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/pool.py", line 223 in _repopulate_pool
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/pool.py", line 159 in __init__
  File "/usr/local/lib/python2.7/site-packages/sklearn/externals/joblib/pool.py", line 402 in __init__
  File "/usr/local/lib/python2.7/site-packages/sklearn/externals/joblib/pool.py", line 563 in __init__
  File "/usr/local/lib/python2.7/site-packages/sklearn/externals/joblib/parallel.py", line 613 in __call__
  File "/usr/local/lib/python2.7/site-packages/mne/connectivity/spectral.py", line 990 in spectral_connectivity
  File "/usr/local/lib/python2.7/site-packages/mne/utils.py", line 551 in verbose
  File "<string>", line 2 in spectral_connectivity
  File "plot_mne_inverse_coherence_epochs.py", line 100 in <module>
```

agramfort, 2015-04-03:
What os? How did you install python ?

breuderink, 2015-04-03:
OSX, homebrew Python. Relevant package versions:
```
$ brew info python
python: stable 2.7.9 (bottled), HEAD

$ pip freeze
numpy==1.9.0
scipy==0.14.0
matplotlib==1.4.0
```

breuderink, 2015-04-03:
I'll try to do a fresh install of Numpy to see if that helps.

breuderink, 2015-04-03:
I switched to Python 3, and now it progresses much further. It can't find the `surfer` package:

```
Traceback (most recent call last):
  File "plot_mne_inverse_coherence_epochs.py", line 118, in <module>
    subjects_dir=subjects_dir)
  File "/usr/local/lib/python3.4/site-packages/mne/source_estimate.py", line 1404, in plot
    colorbar=colorbar)
  File "/usr/local/lib/python3.4/site-packages/mne/viz/_3d.py", line 403, in plot_source_estimates
    import surfer
ImportError: No module named 'surfer'
```
I can't find it with `pip` — where do I get it?

agramfort, 2015-04-03:
They are thread bugs with Apple accelerate. That's why I use anaconda with mkl

Eric89GXL, 2015-04-03:
@breuderink it might be least painful to use Anaconda. I had many problems on OSX trying to use other options, including system python and Enthought. Never tried `brew`, but this issue is making me add it to my list of unsafe options :)

PySurfer is unavailable for Python 3 currently because Mayavi is 2.7 only, and will be for the foreseeable future. I might try to switch PySurfer over to using a different, Py3k-compatible backend at some point, though.

Eric89GXL, 2015-04-03:
Feel free to keep commenting, but I'm going to go ahead and close this as a problem outside our package

closed 2015-04-03 by Eric89GXL

breuderink, 2015-04-03:
Ok I'll try Anaconda then, I see it supports Mayavi as well. Thanks for the help!

amitjaiswal11111, 2016-08-22:
Hi,
I am using MNE python on the platform configuration as following:
Python 2.7.11 |Anaconda custom (64-bit)| (default, Dec  6 2015, 18:08:32)
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux2
when I am running the example function "plot_mne_inverse_psi_visual.py" and also many other functions, it's showing an error "ImportError: No module named surfer". 
As I understand, it's lacking Pysurfer. Or somthong different? How can I install this properly??
Thanks a lot...

agramfort, 2016-08-22:
yes you need to install pysurfer

pip install pysurfer

should do the job