mne-tools/mne-python #1934: Documentations TODOs

agramfort, 2015-04-07:
a WIP list of FAQ / common pitfalls using MNE

- +/- in Evoked
- the use of Montage
- working with a template brain

please expand

Eric89GXL, 2015-04-19:
Also moving this to 0.10 unless @agramfort can bribe someone

agramfort, 2015-04-19:
bribe with what? beers? :)

Eric89GXL, 2015-04-19:
I seem to remember you having an office full of people working with you -- maybe you can convince one of them that creating effective documentation is critical in scientific software development, and therefore important for their training and career development.

On second thought -- yeah, let's go with beers.

agramfort, 2015-04-19:
hehe @mainakjas @yousrabk @teonlamont what do you think? beers or career argument? :)

teonlamont, 2015-04-19:
do we have an intended date for 0.9 release?

Eric89GXL, 2015-04-19:
Planning for Friday or earlier, unless there is a compelling reason to wait. @mainakjas's interpolation would have been one for me, but he seems to have just about finished that. I think the issues and PRs currently tagged as 0.9 can be done by then. Making the docs better could also be a potential reason to delay release if someone is willing to tackle it.

Eric89GXL, 2015-05-09:
Added the idea of having IPython notebooks linked to / in docs from @mainakjas

agramfort, 2015-05-10:
having a way to have people contributing to the doc via tutorials notebooks
and rendering them in the main doc could facilitate documentation
contributions...

Eric89GXL, 2015-05-15:
Rolled this into #1495

closed 2015-05-15 by Eric89GXL