mne-tools/mne-python #1935: list publications which use MNE-Python

mainakjas, 2015-04-07:
We should have a page where we list all the publications that use MNE-Python for their analysis.

Thoughts @teonlamont @agramfort @dengemann @Eric89GXL ?

teonlamont, 2015-04-07:
I think it's a good idea. It will help boost the visibility of the project even more.

dengemann, 2015-04-07:
+1 maybe directly parse scholar citations for the two articles.

agramfort, 2015-04-07:
+1 for automatic parsing

jasmainak, 2016-02-20:
It turns out there is this: https://github.com/ckreibich/scholar.py. So, we can just copy that one file in `externals` and automatically do the parsing. Looks like a nice Sunday hack for someone bored :)

agramfort, 2016-02-22:
it would be nice indeed

jasmainak, 2016-02-26:
I think I (almost) got it working, but the problem is that google scholar thinks the script is a bot (which it is?) -- so it fails half of the time. So, looks like it's a dead end.

Just for pedantic value, I am sharing the [code](https://github.com/mne-tools/mne-python/compare/master...jasmainak:scholar_parser?expand=1). It's basically a hacked version of the [google scholar parser](https://github.com/carlosp420/google_scholar_parser) -- but I changed a bit the `parse_soup_page` function, which parses the html using [Beautiful soup](http://www.crummy.com/software/BeautifulSoup/bs4/doc/)

kingjr, 2016-02-26:
When I had tried parsing gscholar a few years ago, I got banned several times (i.e. can't access the page for 12h)... FWIR Pubmed has an easier and less defensive interface.

kingjr, 2016-02-26:
You can also just create gscholar notifications, and set up a mail server to collect that information.

jasmainak, 2016-02-26:
interesting. Thanks for the ideas. Let me think about it tomorrow at the brainhack

kingjr, 2016-02-26:
(gscholar used to have RSS feeds for notifications, not sure whether this is still the case; if so that should be an easy hack)

jasmainak, 2016-02-26:
one question. Did you also face anti-bot policy with pubmed? Or you didn't try enough to know?

kingjr, 2016-02-26:
Not an explicit one, but FWIR it seemed pretty slow.

dengemann, 2016-02-26:
Maybe this helpful here, a small library for dealing with pubmed i wrote a few years ago.

https://github.com/PyMed/PyMed

It avoids getting blocked and so on.

jasmainak, 2016-02-26:
amazing, looks like I have a task at hand tomorrow :) If you guys want to stick with Google Scholar, looks like I [found a solution](https://github.com/arunpatala/captcha.irctc) ;)

kingjr, 2016-02-26:
Captcha won't last for [long](http://www.wired.com/2014/12/google-one-click-recaptcha/); you'll need to use a convnet...

closed 2016-02-29 by Eric89GXL