mne-tools/mne-python #1938: Bug: AttributeError when ICA analysis is run using the mne version: '0.9.dev0'

Anevar, 2015-04-08:
When running a script for ICA , **AttributeError: 'module' object has no attribute 'isclose'** is generated with the new mne version. Crossed checked with the mne examples for ICA with version '0.9.dev0' and version '0.9.dev'. I was able to run the script with the previous version of the mne: '0.9.dev' without this error. 

 Error is generated at the line:
 ecg_inds, scores = ica.find_bads_ecg(ecg_epochs, method='ctps').

mainakjas, 2015-04-08:
Can you give the full error trace?

Anevar, 2015-04-08:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/spyderlib/widgets/externalshell/sitecustomize.py", line 493, in runfile
    execfile(filename, namespace)
  File "/meg/meg_tmp/erntime_2012/Scripts/preprocess_raw_data.py", line 94, in <module>
    ecg_inds, scores = ica.find_bads_ecg(ecg_epochs, method='ctps')
  File "<string>", line 2, in find_bads_ecg
  File "/home/src/mne/mne/utils.py", line 537, in verbose
    return function(*args, **kwargs)
  File "/home/src/mne/mne/preprocessing/ica.py", line 889, in find_bads_ecg
    ecg, times = _make_ecg(inst, start, stop, verbose)
  File "<string>", line 2, in _make_ecg
  File "/home/src/mne/mne/utils.py", line 537, in verbose
    return function(*args, **kwargs)
  File "/home/src/mne/mne/preprocessing/ecg.py", line 301, in _make_ecg
    ecg = np.hstack(inst.crop(start, stop, copy=True).get_data())
  File "/home/src/mne/mne/epochs.py", line 1224, in crop
    tmask = _time_mask(self.times, tmin, tmax)
  File "/home/src/mne/mne/utils.py", line 1728, in _time_mask
    mask |= np.isclose(times, tmin)
AttributeError: 'module' object has no attribute 'isclose'

dengemann, 2015-04-08:
@Anevar which numpy version do you have installed? What does np.__verion__ return? We might need to backport isclose.

Anevar, 2015-04-08:
I could be wrong when I say this as I'm not sure the error has anything to do with the version of the numpy. 
The ICA example works when I do:
**git checkout f86814f863b31c77ba43b433f03160c017956747 -b v0.9.dev** 

np.version returns me this: *'numpy.version' from '/usr/lib/python2.7/dist-packages/numpy/version.pyc'*

dengemann, 2015-04-08:
@Anevar sorry, github markdown messed it up. I was going to say `np.__version__`

dengemann, 2015-04-08:
And it's clearly a numpy issue.

Anevar, 2015-04-08:
numpy version: '1.6.1'

dengemann, 2015-04-08:
Ok, there we go. Any chance that you can upgrade your numpy version? We can meanwhile backport np.isclose.

Anevar, 2015-04-08:
May take a while. Sorry! But I'll update it and let you know. 
Thanks!

agramfort, 2015-04-08:
I used to backport everything for neurospin old numpy...

@Anevar : option 1 you make the fix locally and send us a proper PR with
backport (see fixes.py)

option 2 : you use Anaconda

1 is the best but the hard way...

eickenberg, 2015-04-08:
We'll do option 1

dengemann, 2015-04-08:
>We'll do option 1

That's the spirit! +1

mainakjas, 2015-04-08:
> We'll do option 1
>
> That's the spirit! +1
@Anevar you have all the github notes :)

closed 2015-04-12 by teonlamont