mne-tools/mne-python #1939: Plot difference waves

teonlamont, 2015-04-08:
There should be a wave to easily plot difference waves from a list of Evokeds. this is type that is widely used in EEG. I think at a minimum it should take picks, a stats function (mean, rms), and an option to invert the axes. something similar is currently implemented with the `mne.viz.plot_topo`.

dengemann, 2015-04-08:
@teonlamont can you give an example? Not sure I'm understanding the feature you're proposing.

teonlamont, 2015-04-08:
something like this:

[image]

dengemann, 2015-04-08:
@teonlamont and what do we see in that plot?

agramfort, 2015-04-08:
can you pass multiple Evoked to plot topo already?

dengemann, 2015-04-08:
> can you pass multiple Evoked to plot topo already?

yes

dengemann, 2015-04-08:
It seems @teonlamont want's something like an SNR/GFP plot for comparing conditions?

teonlamont, 2015-04-08:
> @teonlamont and what do we see in that plot?

this is the averages of the central EEG electrodes in a N400 paradigm: coherent versus anomalous.

dengemann, 2015-04-08:
@teonlamont so you propose an option for aggregating channels before plotting each evoked?

teonlamont, 2015-04-08:
yes, that would be the stats parameter: rms, average, etc.

agramfort, 2015-04-08:
I am not sure to follow. You want to plot z-scored data?

dengemann, 2015-04-08:
@teonlamont I'm wondering if we need an API for that.

```Python
fig = plt.figure()
for evoked, ls in zip(evokeds, ['-', '--']):
    plt.plot(evoked.times, evoked.data.mean(axis=0), linestyle=ls)
```
and so on.

dengemann, 2015-04-08:
The thing why I'm a bit reserved about this idea is that it creates an API for doing many things which can potentially gotten wrong. E.g. compute GFP without taking into account the rank of the data. If it's about averaging over channels, in the sense of an ROI it should only do that I feel. @teonlamont 
What about:

```[e.plot_roi(roi_channels) for e in evokeds]```

dengemann, 2015-04-08:
or some general function like that which takes a list of evokeds.

teonlamont, 2015-04-08:
it's the general practice in ERP research to plot difference waves between two contrasting conditions. I was thinking that it would be a nice feature to be able that. I would save the two functions commonly used would be average for ERPs and rms for ERFs.
I was initially thinking to modify `plot_evoked` to take list of evokeds because it already has the parameters to adjust the plots, and that the waves would all be in one plot. I guess something similar to what you are proposing with `plot_roi`.
`plot_topo` does this already but it does it for each channel.

agramfort, 2015-04-08:
better than

http://martinos.org/mne/dev/auto_examples/plot_topo_compare_conditions.html#example-plot-topo-compare-conditions-py

?

dengemann, 2015-04-08:
@teonlamont my gut feeling tells me that  it will not be that easy to get this really right. We have MEG and EEG and then MEG + SSS with reduced rank. We expose different SNR plots and now we would expose yet another related measure, RMS. We need a very sharp API if we want to go that way.

teonlamont, 2015-04-08:
@agramfort yes that, but averaged over a `pick` of channels, just one plot.

agramfort, 2015-04-08:
I would add a plot global field power function
that can take a list of Evoked

we have an issue on this

dengemann, 2015-04-08:
> I would add a plot global field power function
that can take a list of Evoked

We already have the whitened global field power, the SNR, ...

Second @teonlamont want's not only GFP like plots but averages over channels.

jona-sassenhagen, 2015-04-08:
I have a function that averages over automatically determined or pre-defined ROIs and plots curves + shaded CIs/SEs per condition using seaborn, maybe that'd help. (It's fallout from the viz.plot_topo thing I'm doing.)

agramfort, 2015-04-08:
I am voting +1 for such a function.

teonlamont, 2015-04-19:
converging the conversation to #1276

teonlamont, 2015-04-19:
reopening

Eric89GXL, 2015-04-19:
So what is the API proposal here? (Or in #1276 for that matter)?

jona-sassenhagen, 2015-04-19:
If I'm getting it correctly, the point is combining electrodes, not so much making difference waves, as the latter function is already available as subtracting evokeds.

Maybe like viz.plot_topo? You don't tell it what contrasts to plot, you do that by subtracting the evoked objects by hand, but that's IMO very tidy anyways if you have a list of epochs, or a dict of lists of evokeds.

```
viz.plot_roi(list_of_{evokeds, epochs, tfr), 
             rois={dict of electrode:roi or tuple of (row, col)}, 
             cosmetics ...)
```

teonlamont, 2015-04-19:
sorry, I was speaking about both, plotting difference waves of a particular contrast, but doing it for a selection of sensors, not just a single sensor.

I like the proposal of plot_roi. you could supply a `pick` for the roi or a list of pre-specified rois like you mentioned both, and you can supply a function (mean, RMS), then cosmetics like you said.

jona-sassenhagen, 2015-04-19:
I like confidence bands in addition to, or even instead of, the mean. However, for that, you need at least subject averages (or store them at the time of averaging).
But maybe that's too much work. Also, most people don't plot them anyways cause they're so wide

jona-sassenhagen, 2015-04-19:
I don't really experience a need for giving difference waves any specific status beyond any other evoked object. 
```
conditions = ["congruent", "incongruent"]
evokeds = [dict_of_grand_evoked[cond] for cond in conditions]
mne.viz.plot_topo(evokeds, ...)
``` 
Seems very concise and transparent to me.

teonlamont, 2015-04-19:
ok, maybe I just need to write an example to see if/how this would be done without imposing a new function. I think auto-roi plot would be a cool addition.

agramfort, 2015-04-19:
don't call it roi but selection. I suspect we can avoid a new function and
use picks or selection parameter in the worse case.

jona-sassenhagen, 2015-04-19:
Only if this would be restricted to plotting a single GFP, right? What's rather prevalent in ERP research is plotting the means of multiple ROIs in one plot. So you'd have 30 or 60 electrodes, and plot them as 9 ROIs.

agramfort, 2015-04-19:
so 9 lines in one plot? I think indeed that we need a new function...

Eric89GXL, 2015-04-22:
@teonlamont this one also for 0.10, or do you think you can get it done in a day?

jona-sassenhagen, 2015-04-24:
Here's a function that takes as arguments a dict of {cond_name_1:evoked1, cond_name_2:evoked2, ...} and optional row and column determiners (to establish ROIs). It then assigns sensors to ROIs automatically by checking 2d interpolated positions against a row x column rectangular array, calculates means by condition and ROI and plots them in the row x column array using pandas.
IMO It looks pretty good if you do "import seaborn" first.
It can easily be used to plot difference waves if the dict holds evokeds constructed via evoked1-evoked2.

You can probably turn the mean into an RMS calculation too. However, column_names and roi_names must be iterables, so it will NOT do GFP.

```
def plot_roi(evokeds, 
             column_names=["Left", "Midline", "Right"], 
             row_names=["Frontal", "Central", "Posterior"]):
    # "evokeds" is a dict of Evoked objects. 
    # The dict keys are the condition names. 

    # column_names and row_names can be arbitrary, but 
    # you'll get warnings or errors if they're inappropriate
    # examples:
    # column_names = ["Left", "Left-mid",  "Midline", "Right-mid", "Right"]
    # column_names = ["Left", "Left-mid",  "Right-mid", "Right"]    
    # row_names = ["Frontal", "Central", "Posterior", "Occipital"]

    import pandas as pd
    dfs = pd.concat({cond:evokeds[cond].to_data_frame() 
                     for cond in evokeds.keys()})

    def _area(a, b):
        dx = min(a[0] + a[2], b[0] + b[2]) - max(a[0], b[0])
        dy = min(a[1] + a[3], b[1] + b[3]) - max(a[1], b[1])
        if (dx >= 0) and (dy >= 0):
            return dx * dy
        else:
            return 0

    layout = mne.channels.layout.find_layout(evokeds[list(evokeds.keys())[0]].info)

    columns = len(column_names)
    rows = len(row_names)

    if not columns%2:
        print("Warning: an even number of coronal cuts is requested. "
              "This will lead to strange results if midline electrodes "
              "are used.")

    grid = [((1/columns*col_index), 0+(1/rows*row_index), 1/columns, 1/rows)  
            for row_index in range(rows)
             for col_index in range(columns)]

    # we construct a bunch of dicts to map between electrodes, ROIs, etc and back
    # check these to know what ROI an electrode ended up in
    roidict = {ch_name:np.argmax([_area(a, g) for g in grid]) 
                for ch_name, a in zip(layout.names, layout.pos)}

    roi_names = [col + " " + row for row in reversed(row_names) for col in column_names]

    roi_to_name = {roi:name for roi, name in zip(range(len(roidict)), roi_names)}
    name_to_roi = {name:roi for roi, name in zip(range(len(roidict)), roi_names)}

    from collections import defaultdict
    roi_to_electrode = defaultdict(list)
    for key, value in roidict.items():
        roi_to_electrode[value].append(key)

    if len(roi_to_electrode) < (columns * rows):
        print("Warning: electrode coverage too sparse for requested layout.")

    # now the actual plot ...
    f, splts = plt.subplots(rows, columns, sharex=True, sharey=True)

    for r, p_line in zip(row_names, splts):
        for c, p in zip(column_names, p_line):
            roi = name_to_roi[c + " " + r]
            # this is the call to seaborn
            for cond in evokeds.keys():
                dfs.loc[cond].loc[:,roi_to_electrode[roi]].mean(axis=1).plot(ax=p)
            p.set_title(r + " " + c)
            if row_names[-1] not in r:
                p.set_xlabel('')
            if column_names[0] not in c:
                p.set_ylabel('')
            if c == column_names[-1] and r == row_names[-1]:
                lines = p.get_lines()
                p.legend(lines, evokeds.keys(), loc='lower right')

    rois = [" ".join([roi_to_name[roi], ":", ", ".join(roi_to_electrode[roi])]) 
             for roi in range(len(roi_to_electrode))]

    return f, rois
```

It's a simplified version of what I've sent to @teonlamont before.

I think something like this could be in MNE eventually for the ERP people (like me), at least as an example, but I got my hands full and won't open another PR of this size before I get viz.plot_topo done. If people think it looks useful, I could try it in a few weeks.

Example with dummy data (w/ seaborn):
[image]

agramfort, 2015-04-25:
I would make it a WIP PR or an issue with a link
to gist so we keep track of this. It needs some
API discussion but I agree it should be useful
to some people.

jona-sassenhagen, 2015-05-29:
I'm cleaning up the above code a bit (while I'm struggling to get viz.topo handled ...). Q: it requires Pandas right now. Pandas is afaik a soft dependency. Can I use Pandas for this function, or should I try to do without? If Pandas is okay, I'd clean up the code and make a WIP PR.

dengemann, 2015-05-29:
> I'm cleaning up the above code a bit (while I'm struggling to get viz.topo handled ...). Q: it requires Pandas right now. Pandas is afaik a soft dependency. Can I use Pandas for this function, or should I try to do without? If Pandas is okay, I'd clean up the code and make a WIP PR.

Please try to avoid pandas. We don't want it in our core functions, maintaining pandas is really challanging as they update so often that deprecation cycles are meaningless practically, the pandas code changes too fast to be fun at a library level.

jona-sassenhagen, 2015-05-31:
I now have code w/o pandas. One last question before I make a PR: what module should this be in? mne.viz.topo?

It would include one probably internal helper function for automatically assigning channels to ROIs.

teonlamont, 2015-05-31:
I would say under viz for now.

teonlamont, 2015-05-31:
Is it for evokeds only? if so viz.evoked is probably the right way to go

jona-sassenhagen, 2015-06-01:
@teonlamont : it's for evokeds, but it can hardly be a method of Evoked as it's used for multiple conditions. So I guess a function sitting under viz.evoked is a good idea.

agramfort, 2015-06-01:
sounds good

dengemann, 2015-06-02:
I think all agree here.

Eric89GXL, 2015-10-19:
Closing for updated discussion in #2317

closed 2015-10-19 by Eric89GXL