mne-tools/mne-python #1944: raw.filter preload

teonlamont, 2015-04-10:
Is it currently necessary to have the data be preload to filter the data? If I understand it correctly, the data are being grabbed in chunks, which could be done from disk instead of preload the file.

also, is it feasible to add a copy parameter to the filter method instead you would like to test out multiple filtering settings on your data and not have to construct a raw object separately?

Eric89GXL, 2015-04-10:
Our FIR and IIR filtering are both acausal. You could think about loading the necessary chunk or chunks to get the FIR filtering right, but it would be a big pain, especially since our filtering functions do some zero padding. You couldn't do it at all for IIR filters -- the "I" stands for "infinite" which means in general you can't reduce the effect to a limited number of samples (i.e., the output at the last sample may depend on the very first sample). And the fact that we do forward-backward IIR filtering makes it so that you can't even process in sequential chunks.

For the second point you can just do `raw2 = raw.copy()` followed by `raw2.filter()`. I think using different filter settings is less common than using one so I'm -0.25 on adding a `copy` param, but if others want it I can live with it. If we do have it, the default should be `copy=False`. I guess it would need to return the raw object? Not sure what it returns now, if anything.

teonlamont, 2015-04-10:
ok, those points make sense. I was only thinking about the FFT and FIR.
might be more minuses than pluses for this.

Toward the second point, I forgot that we had the copy method. I don't
think it is necessary to add the parameter. currently, the filtering is
done in place and doesn't return anything.

-teon

Eric89GXL, 2015-04-10:
Don't get me wrong, it would be great if we could have delayed filtering, even if it was only available for the FIR filters. It would likely be a bunch of work to get it right, though. I won't stop someone else from trying :)

teonlamont, 2015-04-10:
yeah, I would too. I will need to give the code a looksy and see what it would require. I always seem to find a way to occupy my weekends... ;)

agramfort, 2015-04-10:
-1 for me here. It would be a real pain / mess and we have other pressing
matters...

Eric89GXL, 2015-04-10:
Like a release? :)

agramfort, 2015-04-10:
for example :)

do you volunteer to take care of it?

Eric89GXL, 2015-04-10:
I won't have time until a week from Monday at the earliest, but if you can wait until then, then I could push to get some of the remaining issues/PRs merged.

agramfort, 2015-04-10:
I accept the offer :)

Eric89GXL, 2015-04-10:
I'm going to close this issue for now, @teonlamont feel free to re-open if you can convince @agramfort with an awesome, simple, robust implementation

jona-sassenhagen, 2015-04-10:
FWIW I do something like this a lot:
```
raw2 = deepcopy(raw)
raw2.filter(1,None)
```
It's very handy for ICA.

agramfort, 2015-04-11:
>
> raw2 = deepcopy(raw)
>
> we recommend to do:

raw2 = raw.copy()

but it should work eitherway

teonlamont, 2015-04-12:
closing for now. may be reopened in the future

closed 2015-04-12 by teonlamont