mne-tools/mne-python #1949: bug: to_data_frame index is truncated to integer rather than being float

chanansh, 2015-04-10:
When I try to convert it to data frame with the as_data_frame I get a warning that it is going to be deprecated but when I use to_data_frame I still get wrong time index.
I would expect here to have a time index with multiplies of 1/256.0 (sampling duration) as I have an rawArray with sampling rate 256.0 (Hz)

Example:

rec.data.as_data_frame().head(10).index
Int64Index([0, 3, 7, 11, 15, 19, 23, 27, 31, 35], dtype='int64')

rec.data.to_data_frame().head(10).index
Converting time column to int64...
Int64Index([0, 0, 0, 0, 0, 0, 0, 0, 0, 0], dtype='int64')

rec.data.info['sfreq']
256.0

Confirmed by Chris Holdgraf: "Hey Hanan - you're right, the index is converted to an integer. I'll take a look at this today to see what may be going on. Would you mind opening an issue on github so we can track progress on this?"

Eric89GXL, 2015-04-10:
@chanansh can you see if this is fixed in `master` now? @choldgraf implemented a fix a couple of hours ago.

Eric89GXL, 2015-04-20:
Any luck @chanansh?

chanansh, 2015-04-20:
sorry didn't have time to check yet.

*HS*

Eric89GXL, 2015-04-20:
I'm going to go ahead and close this -- feel free to ping us if it turns out to still be incorrect

closed 2015-04-20 by Eric89GXL