mne-tools/mne-python #1954: Consolidating list of new functions

teonlamont, 2015-04-12:
reiterating some points @dengemann made in #1703. there was a discussion about a poliferation of functions in dev0.9. some may redundant, e.g. `set_channel_positions` and the Montage workflow: `read_montage` and `apply_montage`. I think we should try to hash out what we need and don't. I think before the release, we should definitively clear these before we have a deprecation nightmare in the future. If we want to keep the particular function even though it is very similar to another one, we should make it private. These are the things that come to mind. 

Montages
Digitizer Data
Referencing

agramfort, 2015-04-13:
sounds fair to me

can you be more specific about some aspects?

Eric89GXL, 2015-04-22:
@teonlamont it looks like you have a lot of remaining issues for release, some of them probably have to get the boot. Move this to 0.10?

dengemann, 2015-04-22:
This one would actually a good reason to delay the release. It's kind of
weird to release stuff of which we know already it's more ore less broken
from a design perspective ... We will have many deprecations soon.

dengemann, 2015-04-22:
The challenge with this is that we need to converge, it's more a design
than a coding job.

2015-04-22 19:18 GMT+02:00 Denis-Alexander Engemann <
denis.engemann@gmail.com>:

Eric89GXL, 2015-04-22:
If it's a release blocker it should be the highest priority. Who needs to discuss/address this? I never really use these functions.

Eric89GXL, 2015-04-22:
And is this related to #1703?

dengemann, 2015-04-22:
So @teonlamont @christianbrodbeck @wmvanvliet @jona-sassenhagen me and @fraimondo are probably the ones who should discuss.

dengemann, 2015-04-22:
I'm now taking a look with @fraimondo and we'll try to summarize our view on this in a bit.

dengemann, 2015-04-22:
I would suggest to:

- making apply_montage private
- add .apply_montage method to our related channels mixin
- make channel pos getters and setters private

christianbrodbeck, 2015-04-22:
Sorry I assigned myself to do this and then got super busy setting up a new experiment and now I'm running subjects for at least a week. While the subjects are in the scanner I can only work on my laptop so not the best place for refactoring...

Just reading the list again I had one thought that Python uses the term "update" for in-place modifications, so maybe you could consider an `.update_montage()` method? Would also be consistent with info being a  dictionary.

dengemann, 2015-04-22:
> Just reading the list again I had one thought that Python uses the term "update" for in-place modifications, so maybe you could consider an .update_montage() method? Would also be consistent with info being a dictionary.

sorry, where does this fit in? I'm missing the discussion on `.update`

christianbrodbeck, 2015-04-22:
There is no existing discussion, I'm just suggesting a name for a method for Info

dengemann, 2015-04-22:
Ahh. So the question would be whether we add a method to info ... `.update_montage` or `.apply_montage` to e.g. Raw, Epochs, Evoked, etc.

fraimondo, 2015-04-22:
From the "end user " point of view it might be more consistent to apply a montage to either Raw,  Epochs, or Evoked objects, without dealing with the concept of Info objects. 
Just think how a new MNE-Python researcher would do to plot a topography.

1. Read data
2. Read montage
3. Apply montage to data
4. Plot

Theres no Info object or attribute mentioned in that very simple example.

jona-sassenhagen, 2015-04-22:
... is there even a need for a public function? I load my Brainvision (EEG) data and set the montage during IO, apply_montage is dealt with internally, and that's it.

christianbrodbeck, 2015-04-22:
Yes I agree @fraimondo. @jona-sassenhagen  the use case would be where you have a fiff file that does not contain the right montage. I guess an alternative would be to add a montage argument to all read_raw functions?

dengemann, 2015-04-22:
>  Would an alternative be to add a montage argument to all read_raw functions?

That can be ruled out, often you would modify names of the montage object or plot it before applying it.

christianbrodbeck, 2015-04-22:
> That can be ruled out, often you would modify names of the montage object or plot it before applying it.

as @jona-sassenhagen suggested you could create, plot and modify the montage object before loading your data. And then use `data = read_raw(..., montage)`. Or is there a case where you want to plot your data and then modify the montage?

fraimondo, 2015-04-22:
How about plotting ICA objects? I might want to change the electrode names to make a special topo.

It might be pretty handy to be able to apply a montage to a custom evoked created out of special data.

agramfort, 2015-04-22:
what would be better/nicer about loading
montage before loading the data from disk?

christianbrodbeck, 2015-04-22:
@agramfort You could reduce 

1. Read data
2. Read montage
3. Apply montage to data
4. Plot

into 

1. Read montage
2. Read data
3. Plot

But if there are use cases where you need to apply the montage after loading the data then that's probably a moot point.

fraimondo, 2015-04-22:
I was thinking about the ipython way of using mne. Most of the time when we
are researching into new methods we don't need the montage until the point
when we want to plot a topo. And I don't really like the idea of re-reading
the data. Since sometimes it might not come from a file, rather from some
analysis, or it might be way to big to do it.

agramfort, 2015-04-22:
to avoid one line of code we need to add a param to the raw reader that for
many use cases will not be needed. So I prefer the 1st option

jona-sassenhagen, 2015-04-22:
Thanks to @teonlamont , read_raw_brainvision already takes a montage keyword and IMO it's quite comfortable. But no big deal either way.

dengemann, 2015-04-22:
Let's please don't flood all the readers with montage keywords. If I can push -f to this debate let's please have an instance method for applying montage objects.

Eric89GXL, 2015-04-22:
Sounds like there are more advantages to having a method than having a reading `kwarg`, so +1 for method from me. I don't think anyone objected too strongly so I think that's decided. @dengemann or @teonlamont do you have a handle on it to implement and document?

Were there other issues? @christianbrodbeck what about #1703?

teonlamont, 2015-04-22:
sorry, just got to the States so I miss a lot of the conversation.

for the montages, all EEG systems have a montage argument. It's been that way for a while now. I reorganized all the readers to have a consistent api.

although you can load the montage at the construction, I am completely with the for group for keeping a public function to apply the montage to the object.

there will be two types of Montages: Montage and DigMontage

the DigMontage follows the logic of having a simple interface for the user to view all their digitizer information which is useful when you are looking at Device-to-Head transformation, and whether the HPI coils are all good (I have had several instances where the marker actually fell off the subject which rendered a really bad `dev_head_t`. this will allow for the inspection and its correction or removal). This closes #1703 and makes for another consistent API usages.

I think the set_channel_positions and the get_channel_positions should be made private. I think if we want to get the channels, there should be a `find_montage` similar to `find_layout` where the returns an object with all the data and can be easily re-applied without dealing with the complexity of the info structure.

teonlamont, 2015-04-22:
also, the DigMontage and Montage both have a plotting method and this can be expanded in the future (0.10 and later), perhaps a stand-alone digitizer plotter for DigMontage similar to `mne.gui.kit2fiff` and it can be made compatible for Montage too.

dengemann, 2015-04-22:
@teonlamont If you're fine with this, I'll just do a minimal PR to implement the instance method and make the setters / getters private. The rest is then for MNE 0.10, no?

teonlamont, 2015-04-22:
@dengemann sounds good to me. that would help clean up a good chunk of things.

teonlamont, 2015-04-22:
wait, which instance method?

dengemann, 2015-04-23:
{Evoked, Epochs, Raw, ICA, AverageTFR}.apply_montage

or set_montage

teonlamont, 2015-04-23:
ah, got it

dengemann, 2015-04-23:
I think I'd prefer set_montage over apply_montage

What do you think folks?

teonlamont, 2015-04-23:
i prefer set_montage to apply_montage.

@christianbrodbeck suggested update_montage.

-teon

dengemann, 2015-04-23:
I'm against update_montage ... ther might not be any montage, set_*  is shorter and more precise.

Eric89GXL, 2015-04-23:
Sounds like two votes for `set_montage`, so just go with that one. @dengemann you're on it?

dengemann, 2015-04-23:
Yep!

dengemann, 2015-04-23:
Looking at the `SetChannelsMixin` we now have quite a few methods, a bundle:

- `get_channel_positions`
- `set_channel_positions`
- `set_channel_types`
- `rename_channels`

this is expert stuff but pretty consistent as a "bundle", maybe we want to keep the set/get channel methods public? I would not see a reason to keep parts of them private and the other parts public ...

christianbrodbeck, 2015-04-23:
I like `set_montage()`, @Eric89GXL what was left to do from #1703 is deciding on a consistent API for handling montages.

teonlamont, 2015-04-23:
how I see the uses of montages:
1. add a function `get_montage` for existing channel info
2. add channel_types attribute to the Montage
3. add `set_channel_types`, `rename_channels` methods to the Montage class
4. rename `apply_montage` to the `inst`

one pitfall i see is that you have to generate the montage to modify channels instead of directly doing so to `inst`. this approach might be opposed to those on dealing with MEG but these things rarely happen for those systems since all of the channel info is set properly. mostly with the interface with EEG and there, you do use a Montage class.

christianbrodbeck, 2015-04-23:
Hmm it seems to me there is a potential for rivalry between Info(/data instance methods) and Montages. What do you do with which? As a naive user I would ask why doesn’t the data just have a `.montage` attribute that is a Montage instance?

dengemann, 2015-04-23:
Why would you ever need get_montage? Can we please keep the Info stupid and simple without custom methods?

agramfort, 2015-04-23:
>
> Why would you ever need get_montage?

+100 please let's avoid YAGNI

+1 for set_montage method

I would make private

   - get_channel_positions
   - set_channel_positions

dengemann, 2015-04-23:
Ok, cool sow we ha our final list. Thanks @agramfort !

Eric89GXL, 2015-04-29:
@teonlamont can we close this one now that your PR is in? If not, do you have time to take care of any remaining issues?

teonlamont, 2015-04-29:
I think this is now done.

closed 2015-04-29 by teonlamont