mne-tools/mne-python #1959: Error in test fieldtrip_client

jmontoyam, 2015-04-12:
mne.\__version__ 
'0.9.dev0'

Steps to reproduce the issue:

1) Download fieldtrip from github.

2) Create environment variable NEUROMAG2FT_ROOT
echo $NEUROMAG2FT_ROOT
/home/user/fieldtrip-master/realtime/src/acquisition/neuromag/bin/x86_64-pc-linux-gnu

3) nosetests -v mne/realtime/

Test fieldtrip_client ... ERROR
Test the RtMockClient. ... ok
Test emulation of realtime data stream. ... ok
Test find_events in rt_epochs. ... ok
Test TCP/IP connection for StimServer <-> StimClient. ... ok

ERROR: Test fieldtrip_client

Traceback (most recent call last):
  File "/home/user/anaconda/envs/mne_env_py2/lib/python2.7/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/user/Documents/Github_Projects/mne-python/mne/utils.py", line 599, in dec
    return function(*args, **kwargs)
  File "/home/user/Documents/Github_Projects/mne-python/mne/realtime/tests/test_fieldtrip_client.py", line 55, in test_fieldtrip_client
    tmax=5, wait_max=1) as rt_client:
  File "/home/user/Documents/Github_Projects/mne-python/mne/realtime/fieldtrip_client.py", line 90, in \__enter__
    raise RuntimeError('Could not connect to FieldTrip Buffer')
RuntimeError: Could not connect to FieldTrip Buffer
-------------------- >> begin captured stdout << ---------------------
FieldTripClient: Waiting for server to start

--------------------- >> end captured stdout << ----------------------

----------------------------------------------------------------------
Ran 5 tests in 5.783s

FAILED (errors=1)

Eric89GXL, 2015-04-12:
Last I knew you needed some beta version of the buffer. @mainakjas is that still true? (Can we get it out of beta?)

jmontoyam, 2015-04-12:
@Eric89GXL thanks for the reply!. I just recently started to configure my development environment because I want to start to contribute to mne-python, and now I'm making sure all tests pass. After checking/installing all the dependencies, I can run all the tests (0 SKIPPED) except the one mentioned in the issue.

Eric89GXL, 2015-04-12:
Do you use the realtime stuff at all? If not, then you could just leave this one skipped for now (our automated build systems will still test it). If you are doing realtime stuff, then we should work harder to figure it out.

agramfort, 2015-04-12:
+1 to skip it

jmontoyam, 2015-04-12:
No, right now I'm not using the realtime stuff, I just wanted to make sure all the tests pass.
I will follow your advice and leave this one skipped for now.
Thanks!

mainakjas, 2015-04-12:
Yep, I can share the beta version with @jmontoyam if he emails me. Otherwise it is safe to ignore the test as it's not inter related to the other things much.

Eric89GXL, 2015-04-20:
Closing for now, since we have now mentioned the possible solutions in case someone else hits this :)

closed 2015-04-20 by Eric89GXL