mne-tools/mne-python #1964: Q: simplification of GAT code?

kingjr, 2015-04-14:
@agramfort @dengemann @cmoutard @jona-sassenhagen @chri4354 @gmeade
While reviewing the GAT code, I found that handling the sklearn decision_function and predict_proba functions significantly complicates the code: notably, it changes the dimensionality of y_pred_, which in turns needs to be handled by the scoring function with some inelegant hacks. 

As this is still in the dev version of MNE, I was wondering whether you'd agree to change a few things:

1. Drop the 'predict_type' option. If users want to use a decision_function or a predict_proba (or anything else we currently don't support) they could easily make their own classifier with a inherited class:
e.g.
```
from sklearn.svm import SVC
class my_SVC(SVC):
    def predict(self, x):
        probas = super(my_SVC, self).predict_proba(x)
        return probas[:,0]
```
2. Drop the automatic definition of the scorer (AUC if predict_type is probabilistic, Accuracy if discrete, Least Square if Regression etc...) and let the user takes care of this by defining the scorer him/herself.

dengemann, 2015-04-14:
@kingjr you're becoming a Python hacker. Let's keep what we have. People who start working with MNE usually don't (want) write their custom classes but might want to use the GAT.

kingjr, 2015-04-14:
@dengemann I agree we should make simple to users. Perhaps we could set by default the typical clf = pipeline(scale, svc.predict_proba + AUC_scorer) while making the GAT structure as simple as possible? 

If they want to use the decision_function, or SVR, they most likely know what they're doing, and should specify the type of scorer themselves. 

Finally, we can make up a notebook on all the cases we think of but are somewhat more advanced than the default parameters.

The changes I propose have a net outcome of -150 lines; just saying ;)

closed 2015-04-26 by dengemann