mne-tools/mne-python #1970: Reading / writing HDF5

choldgraf, 2015-04-15:
Another carryover from the "metadata" issue that was just closed. I've been looking into the read/write hdf5 functionality, and it seems pretty limited right now. I was wondering if it would be useful to expand it a bit.

E.g., from my testing it looks like write_hdf5 breaks if you pass it any MNE-specific objects. It seems like these functions should be smart about MNE files (e.g., if I pass `'{'data': eeg_raw_object}` it would recognize it as an MNE raw object, and store it as `{'data': {'data': raw_array, 'info': raw_info, 'kind': 'mne_raw'}}`. Then, when you read in hdf5 files, it could first check to see if `in_data['data']['kind'] == 'mne_raw'`, and if so it would recreate the `Raw` object from the array and info with `mne.io.RawArray`. Does that make sense?

Alternatively maybe the MNE goals is to keep any timeseries data stored as fiff files where possible?

choldgraf, 2015-04-15:
Actually I saw that the TFR code does something like this:

https://github.com/mne-tools/mne-python/blob/e58532d97488d6c2f84a51185ae8b793ca186097/mne/time_frequency/tfr.py

choldgraf, 2015-04-15:
e.g., something like:

```python
def write_mne_object_to_hdf5(savepath, list_of_objects):
    out = []
    for obj in list_of_objects:
        if isinstance(obj, mne.io.Raw):
            this_out = {'data': obj[:, :][0],
                        'info': obj.info,
                        'mne_type': 'Raw'}
        elif isinstance(obj, mne.Epochs):
            this_out = {'data': obj.get_data,
                        'epochs': obj.epochs,
                        'info': obj.info,
                        'mne_type': 'Epochs'}
        out.append(obj)
        
    write_hdf5(savepath, out)
    
def read_hdf5(loadpath):
    list_of_objects = func_to_load_objs
    for obj in list_of_objects:
        if not obj.has_key('mne_type'):
            continue
        mne_type = obj.pop('mne_type')
        if mne_type == 'Raw':
            mne_obj = mne.io.RawArray(**obj)
```

etc.

Eric89GXL, 2015-04-16:
The HDF5 functions are designed to be as orthogonal as possible to their uses in mne-python functions. In other words, they are meant to only read/write some basic Python types (list, dict with string keys, tuple, str, float, int, and numpy arrays), and how mne-python objects are written and read by them is isolated from that, as you've discovered. I have no problems with adding `HDF5` types for raw, evoked, epochs, or whatever. It's possible that storage requirements or load times might be substantially lowered by doing so. It should be future compatible because the basic read/write methods don't do anything special in the HDF5 sense, and that's a general standard.

So I agree in general about using the functions to potentially write mne-python objects, but I am -1 on making read/write_hdf5 themselves "smarter", since it violates orthogonality.

Eric89GXL, 2015-04-16:
...to avoid violating orthogonality, you could simply rename your current smarter implementation to `read_mne_hdf5` or so, and then it would be fine.

choldgraf, 2015-04-16:
That makes sense. Perhaps it could be a matter of implementing it as a method for individual objects? E.g., the TFR object's `save` method uses hdf5 under the hood. Something similar could exist for the other objects as an option, e.g. `raw.to_hdf5` or `raw.save(type='hdf5')`

So when you use these kinds objects, would you basically just save the data as a fif file, and then save any supplementary information alongside it as an hdf5 file?

Eric89GXL, 2015-04-16:
I think the easiest thing is that if `.save` sees a `.hdf5` extension, it saves out the object in that format instead of FIFF format. `write_hdf5` can write that data plus the info structure together.

choldgraf, 2015-04-16:
That's a good idea - so if write_hdf5 gets a `Raw` object, then it just calls the Raw object's "save" method? Is there a way to "insert" into an HDF5 file right now? E.g., with pandas to_hdf, you specify a file and a key, and you can insert that data into the file that way without overwriting the file.

Eric89GXL, 2015-04-16:
No not quite, again that would violates orthogonality for `write_hdf5` to do anything other than write simple Python objects well.

Inside `raw.save`, it would triage how to save the data depending on file extension. If it were `.hdf5` (or whatever we've been using for hdf5 files), then it does the necessary prep work to just call `write_hdf5` with simple Python objects, as opposed to running the code that writes out a FIFF file. Make sense?

There is no way to insert data into a file currently -- I'm not sure we need to add support for that for now.

Eric89GXL, 2015-04-16:
BTW do you actually want this for saving raw data, or what? I'm not sure whether or not there would be a big advantage over FIFF, and you would lose the ability for have other programs that can currently read raw FIFF files (of which there are many) to do so.

Eric89GXL, 2015-04-16:
We have mostly been planning to use read/write_hdf5 for writing new formats that we create, like the AverageTFR objects, and probably eventually a souped up version of SourceEstimates, allowing for potentially multiple epochs and multiple dimensions (e.g., time-frequency source estimates).

choldgraf, 2015-04-16:
he he - yes I am beginning to think that it's worth re-assessing whether this is worth implementing in the first place :)

I see what you mean about orthogonality and I think you make good points. I'm mostly just trying to figure out the best way to deal with the situation of having both the MNE objects, as well as my own objects with metadata that doesn't fit well into MNE objects. I was thinking hdf5 would be nice because I can save everything into a single file, rather than having to store all data in, e.g., `data_raw.fif` and `data_meta.h5` files. If I have a folder with lots of subjects in it, I could see that getting confusing and clunky. But I'm getting that feeling that I'm being a bit OCD about this...

choldgraf, 2015-04-16:
(btw, if you have any resources on how to read fiff files in matlab, I'd love to see them...I'm trying to get folks in my lab to start storing ECoG files with `.fiff`, because right now they just store them as `.mat` files which makes it a PITA for people using python. However, I haven't found a simple matlab function like `read_raw_fiff(file_path)`, which I think would be necessary for anyone to actually use it.

Eric89GXL, 2015-04-16:
Well if it would really help your use case, it would be backward compatible to have a `root` parameter (kind of like pandas) for the `read_hdf5` / `write_hdf5` functions. Right now they assume they should write to the entire file, i.e. having `root='/'`. I don't think it would be too many lines for you to change it so that a different root like `/subj_name/` could be used instead. At least then you could more arbitrarily choose how to store data in one or multiple `.h5` files, and you could use `read/write_hdf5` to do the heavy lifting for you. I think that's a better option than trying to force data into the FIFF format, or than making another output format for raw data that only mne-python can read easily.

Eric89GXL, 2015-04-16:
You must be able to read/write raw data in MATLAB, since programs like FieldTrip do it. I'd look there. I know there is some MNE-MATLAB code that can do some things, but I'm not sure if it's maintained as much as FieldTrip.

choldgraf, 2015-04-16:
For me, it's about being able to write to the same `.h5` file with two different calls. E.g., I'd do `raw.save('sub5_raw.h5', key='data')` then do 'meta_dataframe.to_hdf('sub5_raw.h5', 'meta')`
or something like this.

And thanks for the tip on matlab, I'll look into FieldTrip...just haven't had much experience with it as I'm not a huge fan of matlab :P

Eric89GXL, 2015-04-16:
I am -1 on writing raw data to a HDF5 file for now (it breaks compat. with a lot of programs to do this), but if you *really* want it, you should be able to use `write_hdf5` to write the `raw._data` and `raw.info`, and then you can use `RawArray` to reconstruct it. I think that, combined with adding a `root` parameter to `read/write_hdf5` will get you what you want.

dgwakeman, 2015-04-16:
the MNE matlab toolbox is used by both SPM and Fieldtrip to interact with fiff files, but I don't think their support is very good for writing fiff files. They rely on their own file format once you read it (last I checked). If you want to write fiff files, check the MNE matlab toolbox: particularly mne_ex_read_write_raw (you may want to make some simple modifications for them and use that).

choldgraf, 2015-04-16:
OK - I will hold off on working on anything super `hdf5` related right now. I'll figure out my own system for saving/storing files.

Regarding matlab, I think it's fine if all people can do is read it in. Nobody else uses `.fif` files, but right now after ecog data is collected, it's often stored as a set of separate `.mat` files, one for each channel, each with its own little metadata structure in it. This kind of kills me inside and I just want things to be stored more clearly/efficiently :)

Eric89GXL, 2015-12-12:
HDF5 output has been put into the `h5io` package, so I'm closing this mne-specific writing issue

closed 2015-12-12 by Eric89GXL

choldgraf, 2015-12-12:
Cool