mne-tools/mne-python #1971: SVD error during PCA computations

kambysese, 2015-04-17:
I have a large dataset on which scipy linalg `decomp_svd` sporadically throws a `LinAlgError("SVD did not converge")`. I can't exactly pinpoint the source of the error. At this time I can only resolve the issue by changing the the sample freq (e.g., 200, , 500, 600, 800, 1000) for calculating projectors. As a result I am unable to use the same sample frequency across subjects, and I can't see a reason to toss the data. Given the frequency the error occurs, I'd be left with very little data to work with if I did. Does any one know what this is and how to systematically deal with it?

Eric89GXL, 2015-04-17:
Can you try updating scipy and numpy and see if it goes away? IIRC this is a known bug at their end.

kambysese, 2015-06-02:
The updates seem to help but not completely solve the issue. Less frequently having to change the sample freq for calculating projectors. Closing this .

closed 2015-06-02 by kambysese