mne-tools/mne-python #1973: Q: weird trans realignment?

kingjr, 2015-04-17:
I have a strange head-position vizualization for one subject.

After tsss the data with maxfilter using a common head realignment to the most representative run (there's one HPÏ measurement at each run), I seem to have no problem with the mne-coregistration of digitizer points.

[image]

However, in mne-python, using ```mne.viz.plot_trans``` gives me the following figure. Not only the head position is extremely bad, but the digitized points are not aligned on the scalp:

[image]

I doubt that this is a script problem, because I use the same for all my subjects.

Has anyone got any idea how I could troubleshoot this?

Eric89GXL, 2015-04-17:
What was your actual call to `plot_trans`? Did you provide the correct trans name?

I don't see any dig points in your mne_analyze plot, can you turn them on for comparison? Those don't look so far off, they might look similar in mne_analyze.

Note that the coreg you do in mne_analyze is independent of what Maxfilter changes. mne_analyze sets the mri_head_t, the transform between the MRI and head coords. MF changes the dev_head_t, the relationship between the MEG and the head. As such, the coreg step says where the red dots are relative to the head, dev_head_t says where the head is inside the helmet. Something to keep in mind when discussing and thinking about this.

Eric89GXL, 2015-04-17:
Note that you can also load each run of the non-Maxfiltered data and use the info from that to plot the trans, since you're effectively only changing the dev_head_t when you do that (the dig, mri_head_t, and head surface will not be changed). It would be good to take a look at those yourself to see if there is variability.

kingjr, 2015-04-17:
> What was your actual call to plot_trans? 

```
scene = plot_trans(raw.info, trans=trans_fname, subject=subject, subjects_dir=op.join(data_path, 'subjects'), source='head')
```

> Did you provide the correct trans name?

Yes, I did check that a number of times.

> I don't see any dig points in your mne_analyze plot, can you turn them on for comparison? Those don't look so far off, they might look similar in mne_analyze.

Do you mean those? If so, they're 'on'.
[image]

By eye, they seem very much aligned, but because they get more invisible with increased precision, it may be hard to detect?

> Note that the coreg you do in mne_analyze is independent of what Maxfilter changes. mne_analyze sets the mri_head_t, the transform between the MRI and head coords. MF changes the dev_head_t, the [...]

Thanks for the clarification. 

I'll try dig further...

Eric89GXL, 2015-04-17:
In addition to my comment above about trying each of the non-MF runs in plot_trans, you could also try intentionally making the coreg worse, e.g. by moving it up a couple of centimeters manually, saving it, and plotting that. Should only take a couple of minutes, and it should change in both mne_analyze and in plot_trans in consistent ways. If it doesn't, it's informative.

kingjr, 2015-04-17:
> you could also try intentionally making the coreg worse, e.g. by moving it up a couple of centimeters manually, saving it, and plotting that. Should only take a couple of minutes, and it should change in both mne_analyze and in plot_trans in consistent ways.

When saving an intentionally bad trans aligment, I get consistent misalignments in mne-python plot_trans.

Eric89GXL, 2015-04-17:
I am still a bit surprised you can only see one or two of the dig points in mne_analyze. Is the coreg really that tight? Or did you omit some points during the process?

Eric89GXL, 2015-04-17:
...I wonder if somehow mne_analyze is loading a different head surface file than mne-python is loading. That is weird but a possiblity. You could chekc the mne_analyze output, it might say what it loads. If not, you could mis-name the file you think it's loading to see if it breaks it.

kingjr, 2015-04-17:
> I am still a bit surprised you can only see one or two of the dig points in mne_analyze. Is the coreg really that tight? Or did you omit some points during the process?

I did not explicitly omit anything. Is there a way to change the color/transparency so that we make sure it's correct? I just check with the MRI viewer instead, which looked fine (see first figure).

kingjr, 2015-04-17:
> ..I wonder if somehow mne_analyze is loading a different head surface file than mne-python is loading. That is weird but a possiblity. You could chekc the mne_analyze output, it might say what it loads. If not, you could mis-name the file you think it's loading to see if it breaks it.

I thought you had a point because the surf file was a symbolic link, but changing it to a real file didn't change a thing; and changing its name breaks both mne_analyze and mne.plot_trans

I had to correct for HPI ordering with maxfilter as a first step on this subject (as well on two other). Could it be something related to this?

Eric89GXL, 2015-04-17:
In mne_analyze the points just get smaller when they're closer to the surface, red outside blue inside. I don't think you can change it. I'm just surprised there are so few points visible. I usually use mne_analyze and the viewer from within that to visualize, is that not what you're doing?

Regarding dig order, that will only change potentally the dev_head_t (where the head is related to the helmet), not the head_mri_t (where the dig points are relative to the MRI head surface). So I guess the question is, are you worried about the red dots in relation to the MRI, or the head in relation to the MEG?

If it's the head related to the MEG helmet, I'd really try using the info from the non-maxfiltered runs to do the plot_trans (this will change the dev_head_t), which I don't think you've done yet...

If it's the red dots, can you try looking in mne_analyze and shift the trans incrementally to make sure the points are actually disappearing because they are directly on the surface?

kingjr, 2015-04-17:
Thanks a lot for your support @Eric89GXL 

> In mne_analyze the points just get smaller when they're closer to the surface, red outside blue inside. I don't think you can change it. I'm just surprised there are so few points visible. I usually use mne_analyze and the viewer from within that to visualize, is that not what you're doing?

Yes, it is.

> Regarding dig order, that will only change potentally the dev_head_t (where the head is related to the helmet), not the head_mri_t (where the dig points are relative to the MRI head surface). So I [...]

This was a bad subject who had to wear plastic glasses that don't fit well in the MEG helmet. So I'm not sure. The current realigned head position seems crazy, but who knows. What I find curious is why the red dots aren't aligned to the head.

> If it's the head related to the MEG helmet, I'd really try using the info from the non-maxfiltered runs to do the plot_trans (this will change the dev_head_t), which I don't think you've done yet...

I've tried on the sss (not tsss) files, and the same misalignement between the head and the red dots. There are also runs in which the red dots aren't present, for a reason that I'm currently digging...

> If it's the red dots, can you try looking in mne_analyze and shift the trans incrementally to make sure the points are actually disappearing because they are directly on the surface?

Yes, I did that. They do disappear in the expected way.

Thanks again!

Eric89GXL, 2015-04-17:
If the SSS'ed files had already been transformed (probably had), then it wouldn't be any different from using the tSSS'ed data, you have to go back to the actual original raw file unfortunately.

kingjr, 2015-04-17:
Ok. I'll get back to you once I download the data.

Eric89GXL, 2015-04-19:
BTW did you have cHPI on during your recording? I have a pull request working on estimating head positions in Python #1876, and I wouldn't mind getting my hands on another site's data (deidentified of course) to benchmark / compare against Maxfliter.

kingjr, 2015-04-20:
No , it was estimated at the beginning of each 5 min run.

agramfort, 2015-04-20:
does it look good in mne_analyze?

mainakjas, 2015-04-20:
I don't have any problem with the somato data which is SSS-ed.

kingjr, 2015-04-20:
@agramfort yes

@mainakjas Note that I don't have any problem with my 16 other (tSSSed) subjects either.

agramfort, 2015-04-20:
I am clueless then...

Eric89GXL, 2015-05-10:
@kingjr did anything ever come of this one?

kingjr, 2015-05-10:
Nope, never found the issue. I'll probably give it another try in 10 days.

Eric89GXL, 2015-12-12:
I'm going to close this but feel free to reopen if you have some more information

closed 2015-12-12 by Eric89GXL

kingjr, 2015-12-13:
OK. I   have left the buggy dataset aside ; I'll reopen the issue if I come
back to it.