mne-tools/mne-python #1983: txt to fif format transformation of eeg data

atalanti, 2015-04-19:
Dear mne-python users,

Some months ago I managed to turn a txt file with my eeg data into fif format using mne-python. I had used the Rawarray class and the create_info from the meas_info.py.

Now I have to use the improved mne-python version. 
I downloaded the latest whole updated mne-python-dev_head_t package from Github, which has a lot of changes in some functions from the one I had.  And now I tried to transform my txt original data to .fif format using again the create_info and the Rawarray class. However, I have some trouble using Rawarray class (i have copied it inside meas_info.py)

my code for the format transformation is this one:
ch_names=('Fc6','Fc2','F8','F4','Fz','AFz','Fp2','Fpz','Fp1','P3','T7','A1','Fc1','Fc5','F3','F5','P8','P4','Cp6','Cp2','A2','T8','C4','O2','Oz','O1','Pz','C2','P7','Cp1','Cp5','C3');

ch_types=('eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg')
sfreq=1000
info=meas_info.create_info(ch_names,sfreq,ch_types)
data=np.load('subject1_data.npy')
myraw=meas_info.RawArray(data,info)
print(myraw._data)
myraw.save('subj1data_raw.fif')

(I have done first .txt -> .mat -> npy)

the error that I get is:

Creating RawArray with float64 data, n_channels=32, n_times=627000
Traceback (most recent call last):
  File "main123.py", line 39, in <module>
    myraw=meas_info.RawArray(data,info)
  File "<string>", line 2, in __init__
  File "/home/atakoulini/Documents/mne-python-dev_head_t/mne/utils.py", line 541, in verbose
    return function(*args, **kwargs)
  File "/home/atakoulini/Documents/mne-python-dev_head_t/mne/io/meas_info.py", line 1221, in __init__
    self.proj = None
AttributeError: can't set attribute

Rawarray uses the Baseraw wich is inside the base.py . I checked the Baseraw into the previous mne-python package that I had and there are plenty of differences with the Baseraw into the mne-python-dev_head_t .

Could you please tell me a way to do the txt to fif transformation of my eeg data?  (My txt file has 627000 raws and 32 lines standing for the 32 electrodes)

thanks a lot in advance,
ata

dengemann, 2015-04-19:
Hello @atalanti it is not a good idea to modify the source code the way you did it. This may lead to some funky interactions and makes it very difficult to help you, technicallt you're now using a version of MNE nobody else uses. This being said, could you possibly share some parts of your data to allow us to reproduce your problem? I don't really see why this script should not work.

atalanti, 2015-04-19:
Hi Denis,
Unfortunately I cannot share the eeg data. However I created a numpy array with smaller dimensions than my data numpy (32 lines x 5700 raws) and random values. I attach it here.  I need to do the txt to fif transformation with the latest version of mne-python-dev_head_t .
Thank you,atalanti

Date: Sun, 19 Apr 2015 12:03:25 -0700

Eric89GXL, 2015-04-19:
@atalanti can you try updating your code (all of it) to latest `master`? You don't need to use mne-python-dev_head_t, because that has now been merged.

atalanti, 2015-04-20:
I updated to latest master. Then I tried again to convert my npy data to fif and still I get the same error which is caused due to Rawarray.

/usr/lib/python2.7/dist-packages/sklearn/utils/__init__.py:11: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from .murmurhash import murmurhash3_32
/usr/lib/python2.7/dist-packages/sklearn/utils/sparsetools/__init__.py:3: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from ._min_spanning_tree import minimum_spanning_tree
/usr/lib/python2.7/dist-packages/sklearn/utils/sparsetools/_graph_validation.py:5: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from ._graph_tools import csgraph_to_dense, csgraph_from_dense,\
/usr/lib/python2.7/dist-packages/sklearn/utils/sparsetools/__init__.py:4: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from ._traversal import connected_components
/usr/lib/python2.7/dist-packages/sklearn/utils/extmath.py:14: RuntimeWarning: numpy.ufunc size changed, may indicate binary incompatibility
  from ._logistic_sigmoid import _log_logistic_sigmoid
Creating RawArray with float64 data, n_channels=32, n_times=627000
Traceback (most recent call last):
  File "main123.py", line 37, in <module>
    myraw=RawArray(data,info)
  File "<string>", line 2, in __init__
  File "/home/atakoulini/Documents/mne-python/mne/utils.py", line 541, in verbose
    return function(*args, **kwargs)
  File "/home/atakoulini/Documents/mne-python/create_info.py", line 189, in __init__
    self.proj = None
AttributeError: can't set attribute

 Is there any way to create my fif file with inputs 1. my npy eeg data, 2. the names of my channels, 3. the types of my channels (eeg) and 4. the  MEG device-to-head transform initialization (since I do not have it at all) , without using the Rawarray? Rawarray does not exist anymore in the patest versions.
 
Thank you,
atalanti

Eric89GXL, 2015-04-20:
Does this work for you?
```
import mne
import numpy as np
data = np.zeros((30, 1000))
ch_names = [str(x) for x in range(30)]
info = mne.create_info(ch_names, 1000., 'eeg')
raw = mne.io.RawArray(data, info)
```
If it does not, you probably do not have an up-to-date installation. If it does, then it should work for your data...

atalanti, 2015-04-20:
It worked. Thanks a lot of time and help.

Eric89GXL, 2015-04-20:
No problem -- let us know if you hit any other snags.

closed 2015-04-20 by Eric89GXL