mne-tools/mne-python #1986: FIX: some TFR examples seem broken on master

dengemann, 2015-04-20:
stockwell and multitaper.

cc @agramfort @Eric89GXL

dengemann, 2015-04-20:
No idea what's wrong here ... might be a blocker. We need to understand what's wrong.

Eric89GXL, 2015-04-20:
Yeah we should definitely make sure all examples work and the doc builds, and not release until they do so. Hopefully we can take care of this and #1780 in the same pass.

agramfort, 2015-04-20:
broken? crash or weird output?

dengemann, 2015-04-20:
The outputs look wrong, see dev exapmles on webpage.

agramfort, 2015-04-22:
we have a blocker here. Something went wrong maybe when refactoring
baselining or something

Eric89GXL, 2015-04-23:
Any idea when it last looked okay? What is it supposed to look like? Is it supposed to look like the non-dev-docs version? If so, I can probably find time tomorrow to bisect and find the problem.

dengemann, 2015-04-23:
@Eric89GXL maybe before your cuda refacoting? Also didn't you also recently refactor some of this code. Maybe let's git praise (blame).

Eric89GXL, 2015-04-23:
Can you describe briefly what it's supposed to look like?

dengemann, 2015-04-23:
[image]

for the stockwell roughly

dengemann, 2015-04-23:
I think this was one original screenshot from the example:

[image]

dengemann, 2015-04-23:
Here some info on the original MTM implementation, also plots.

https://github.com/mne-tools/mne-python/pull/1639

Eric89GXL, 2015-04-27:
bisected the offending commit, naturally it's mine :)

https://github.com/dengemann/mne-python/commit/a19a954b3fbd5a38f5659ba5d294064129c2ea0a

working on it now

dengemann, 2015-05-05:
We're not good yet ... TF examples are stil broken down here.

Eric89GXL, 2015-05-05:
This is what I get for stockwell...?
[image]

[image]

dengemann, 2015-05-05:
Damn it. I don't get that.
What about multi-taper?

dengemann, 2015-05-05:
rebuilding after cleaning all.

2015-05-06 0:58 GMT+02:00 Denis-Alexander Engemann <denis.engemann@gmail.com

Eric89GXL, 2015-05-05:
plot_time_frequency_multitaper_sensors I assume you mean:

[image]

[image]

dengemann, 2015-05-05:
we're good. apologies for the nois, a make clean wa missing.

PR #2031 (2015-04-27): MRG: Revert default
Closes #1986.

@dengemann this seems to fix the examples. The default `dB=False` was changed to `dB=True`. In some places `dB=True` is default (plotting PSD for raw or epochs) and in others it is `dB=False` (plotting TFRs), but I guess that is okay.
closed 2015-05-05 by dengemann