mne-tools/mne-python #1989: ENH: future API development for cov emthods and behavior of 'auto' option

dengemann, 2015-04-20:
To be discussed.

I'd like to get rid of `regularize` in our examples. We could do this by shipping a precomputed regularized covariance.

The second thing to consider is the fine tuning of 'auto'. When would it include more than 'shrunk' and 'empirical', and so on.

Finally it would be nice to have an example that shows the impact of the noise cov on the leadfield, similar to SSS sensitivity.

Eric89GXL, 2015-04-20:
For 0.9 or 0.10?

dengemann, 2015-04-20:
... we need to at least act upon the `regularize` instances. Either exaplain why they are used or implement my proposed fix for 0.9. The rest might be rather 0.10

Eric89GXL, 2015-05-04:
If we deprecate `regularize` then we need to allow `compute_raw_data_covariance` to accept a `method` parameter as well...

dengemann, 2015-05-04:
Not necessariyl, I'm not sure you would need to regularize if you have 50k samples.

dengemann, 2015-05-04:
The difference will be neglegible.

Eric89GXL, 2015-05-04:
Ahh, okay. I guess that would mean even a 1-minute empty room @ 1 kHz would be enough, assuming enough epochs are kept. Maybe we just update the docstring of `compute_raw_data_covariance` to say that you shouldn't need to regularize in most cases...?

dengemann, 2015-05-04:
Yep.

dengemann, 2015-05-04:
And on top of that you only have grads and mags.

dengemann, 2015-05-04:
We can see later if a method param could make sense.

agramfort, 2015-05-05:
sounds like a plan

Eric89GXL, 2015-05-05:
@dengemann would it be enough for now to replace the `cov` w/regularize in examples with a new cov comptutation that just uses `shrunk`?

Eric89GXL, 2015-05-05:
...that seems to add ~50 seconds of computation on my machine just for using `method='shrunk'`, so maybe it's not such a great plan.

Eric89GXL, 2016-03-16:
I'm going to close this, but feel free to reopen if anyone feels we should reconsider `mode` option for `compute_covariance`

closed 2016-03-16 by Eric89GXL