mne-tools/mne-python #1996: Report, toggle mode

teonlamont, 2015-04-21:
It would be nice to have the behavior of the toggle buttons changed. Instead of being subtracting from the initial state, it should be additive after first click.

init: all the buttons are toggled on.
behavior change: click a button, and it selects the given button.
-click other buttons to add additional sections.

I think it is the case that users typically want to focus on a particular section when it is clicked. currently you have to remove all section but the one you're interested.

dengemann, 2015-04-21:
Yeah something is not really good with this. But then I also do not want to click any thing to see a figure.

Eric89GXL, 2015-04-21:
We could have a button that showed all or hid all, then it would only be two clicks to get your isolated plot

dengemann, 2015-04-21:
I was thinking the same. That would be nice.

Eric89GXL, 2015-04-21:
Might be safer to have two buttons, `Show all` and `Hide all` just to be clear about what happens in all situations (e.g., if it were one button and half the things were shown, it's unclear what the behavior should be)

teonlamont, 2015-04-21:
yeah, that was my initial idea but we were brainstorming ways without add'l buttons. I am totally +1 for the additional buttons.

agramfort, 2015-04-21:
bootstrap has nice ways to make tooltips and dropdown menus if we start to
have too many buttons...

mainakjas, 2015-04-21:
@teonlamont gave up so fast? ;)

First question, why would you want to see 3 out of 7 sections if all of them are in the report? I don't understand the reason.

If I understand correctly, we need a faster way to navigate to certain sections right? how about a collapsible section wise table of contents? Like this: http://www.w3schools.com/jquerymobile/tryit.asp?filename=tryjqmob_lists_collapsible. We probably don't even need the buttons on top then (I know people like features that are already there but just throwing ideas :)

teonlamont, 2015-04-21:
@mainakjas I won't have google to teach me javascript on the plane. for the number of buttons, it would be a matter of focusing on particular sections in the report. are you sure you don't want to make that for 0.9, imagine how accomplished you would feel after it was released knowing the world could toggle on and off as many buttons as it would like ;)

mainakjas, 2015-04-21:
> @mainakjas <https://github.com/mainakjas> I won't have google to teach me
> javascript on the plane.
:) Enjoy your trip!

> for the number of buttons, it would be a matter of focusing on particular
> sections in the report. are you sure you don't want to make that for 0.9,
> imagine how accomplished you would feel after it was released knowing the
> world could toggle on and off as many buttons as it would like ;)
I'll let @agramfort decide that :)

agramfort, 2015-04-22:
not for 0.9

teonlamont, 2015-06-16:
@jaeilepp if you need a break from epochs.plot, and are bored, here's something cool to work on ;)

dengemann, 2015-06-16:
@teonlamont I think the butterflies are really exciting ;)

mainakjas, 2015-07-13:
what was the consensus here? what do we want? do we want to keep things as it is or do we want some changes?

closed 2015-07-13 by agramfort