mne-tools/mne-python #2000: ICA label inconsistency

teonlamont, 2015-04-21:
There's an inconsistency in how the ICs are labeled in the `ica.plot_components` (index starts at 0) and how they labels when you use `ica.get_sources` (index starts at 1). Is there a reasoning behind this? This was driving me crazy for a bit when I was computing the time course to the topography.

teonlamont, 2015-04-24:
@agramfort should we keep this behavior?

dengemann, 2015-04-24:
??? Really?

teonlamont, 2015-04-24:
https://www.dropbox.com/s/i7eotj4ljhq7sg3/A0129_OLDT1_ica-report.html?dl=0

agramfort, 2015-04-24:
really ??? ** 2

dengemann, 2015-04-24:
Ah @teonlamont  you mean the channel names ... Yes, it's like with raw data, channel index 0 : MEG001

teonlamont, 2015-04-24:
then shouldn't the IC in the topomap have the same name as the channels, it makes it really inconsistent. if i am looking at the time course to determine say, the ecg component, I could easily exclude the wrong IC because of it by being one off.

dengemann, 2015-04-24:
> if i am looking at the time course to determine say, the ecg component, I could easily exclude the wrong IC because of it by being one off.

just don't select things manually ;)

dengemann, 2015-04-24:
Not sure, I think it's the same with regular channels right? In the other plots we use the indices, but plot.raw shows channel names. And we never had a channel name with 000.

teonlamont, 2015-04-24:
> just don't select things manually ;)

lol, unfortunately no ecg channel.

with regular channels you have to specify the channel name, not an index. I feel that the exclude behavior should be the IC name, starting at one `ica.exclude=['IC 001']`  rather than the index, which would mean, the topo plots should start with one. that would be consistent to how we mark bad channels `raw.info['bads']  = ['MEG 001']`

dengemann, 2015-04-24:
> lol, unfortunately no ecg channel.

but you have magnetometers, ICA will compute a nice ECG from that internally

dengemann, 2015-04-24:
> I feel that the exclude behavior should be the IC name, starting at one ica.exclude=['IC 001'] rather than the index, which would mean, the topo plots should start with one. that would be consistent to how we mark bad channels raw.info['bads'] = ['MEG 001']

please don't do this to me.

dengemann, 2015-04-24:
> that would be consistent to how we mark bad channels raw.info['bads'] = ['MEG 001']

It would really be a pain to make it nam-based inside the ICA code. You really should not select ICA components manually for other than exporative purposes.

teonlamont, 2015-04-24:
> but you have magnetometers, ICA will compute a nice ECG from that internally

yes, but I would remove the wrong component if I were looking at the ic time sources which is marked `IC 003` but has the topography of `IC #002`.

> It would really be a pain to make it nam-based inside the ICA code. You really should not select ICA components manually for other than exporative purposes.

but then the selection of the heart beat is circular if you don't have an ecg channel to correlate with.

dengemann, 2015-04-24:
> but then the selection of the heart beat is circular if you don't have an ecg channel to correlate with.

can we talk? hangout?

teonlamont, 2015-04-24:
yeah, sure :)

dengemann, 2015-04-24:
So maybe we can relax our channel naming here a bit and call on exporting the ICA time courses into Evoked,Epochs,Raw match the channel names with the indices.

I'd not be opposed. cc @agramfort @fraimondo

teonlamont, 2015-09-08:
I think this is now resolved with the new plotting

closed 2015-09-08 by teonlamont