mne-tools/mne-python #2002: Extrapolated MEG field on scalp surface

BushraR, 2015-04-22:
Is there a possibility to get projected field maps on scalp surface in MNE-python
make_field_map currently is projecting MEG field maps on helmet surface only

Thank you

agramfort, 2015-04-22:
As far as I know not yet sorry

BushraR, 2015-04-22:
in the file 
mne-python/mne/forward/_field_interpolation.py
in the definition of  make_field_map

on line 383          _make_surface_mapping(evoked.info, this_surf, this_type,
                              trans, n_jobs=n_jobs)
Do you think if this function is called with head surface instead of helmet surface for MEG evoked data, then that would solve my problem and give extrapolated MEG fields to scalp instead of helmet?

agramfort, 2015-04-22:
that's worth trying !

yousrabk, 2015-04-22:
Yes actually I was using this function to plot the MEG fields to the scalp, you just need to give the surface you want and it will do a re-mapping to that surf.

Do you need the code?
Or I can open a PR with the changes, if we want to add this option the make_field_map.

mainakjas, 2015-04-22:
beware though with using a private function. There's no guarantee that we will support it or that it will be available in future `mne` versions. It can be a temporary workaround for now. The more permanent solution is what yousra suggests with a PR

BushraR, 2015-04-22:
It would be great if I can have a code even if it is a temporary workaround. I need those extrapolated values for my current analysis. Thank you

yousrabk, 2015-04-22:
@BushraR hope the PR #2005 is doing what you wanted.

BushraR, 2015-04-23:
Thank you so much

PR #2005 (2015-04-25): ENH be able to plot MEG fields to the scalp surf
@BushraR this will give you the MEG fields in the scalp if you put to True the head_surf. It will do a remapping to the scalp surf instead of the helmet one. I put in the example how you can use it (commented at the end)

I was using it for my own plots. I don't know if @agramfort @Eric89GXL find this kind of useful to keep it.

Closes #2002.
closed 2015-04-25 by Eric89GXL