mne-tools/mne-python #2004: Cross-condition generalisation w/GAT

jona-sassenhagen, 2015-04-22:
Is there a simple way to do cross-condition generalisation with GAT?

My scripts for dealing with this look rather abysmal, with lots of confusing passages where errors can occur. A prime concern is avoiding circularity for not completely orthogonal designs.

dengemann, 2015-04-22:
Take a look at the related example :)

http://martinos.org/mne/dev/auto_examples/decoding/plot_decoding_time_generalization_conditions.html

dengemann, 2015-04-26:
@jona-sassenhagen ok? Can we close this?

jona-sassenhagen, 2015-04-29:
Sorry ...
My problem is specifically with cross-generalisation where one trial could appear in both validation and testing. For example, we want to train on A vs B, and generalize to A vs C; for this, I'm selecting non overlapping subsets of A for each round.
But maybe that's just an inherently messy task.

dengemann, 2015-04-29:
But this is supposed to work with this API. @cmoutard you're doing this, not?

cmoutard, 2015-04-29:
Not anymore and never with the last API, but I will need to come back to this at some point.

kingjr, 2015-04-30:
If you're using two groups of A (A1 and A2) this is easy, you just have to use a classic generalization across conditions.

Else, it's a bit tricky (even from a theoretical point of view). Have a look at this example I made for @cmoutard 
https://gist.github.com/kingjr/b67bad3136f80fe7b81b

kingjr, 2015-04-30:
The general idea is that you have to 
1. fit with a cross val A versus B,
2. separately predict A (with cross val) and C (with mean-prediction)
3. score A and C predictions

Note that with a mean-prediction the C predictions becomes more stable... So don't do any stats involving some variance tests (across trials).

kingjr, 2015-04-30:
Also have a look at this to predict or score subsets of data
https://gist.github.com/kingjr/6a4cbfb51a5d7da6850a

kingjr, 2015-05-07:
@jona-sassenhagen Did it solve your problem?

jona-sassenhagen, 2015-05-08:
@kingjr thanks, I think that's basically what I wanted! Can we leave this open for a few more days in case I run into any other issues?

jona-sassenhagen, 2015-06-01:
@kingjr sorry I'm stuck again. Currently, what I'm doing is (following the gists):

- gat.fit on A vs. B
- gat.predict A, B, and (with 'mean-prediction') C
- plot, separately for each condition, the ratio of correct predictions

I get huge baseline effects. For example, if I plot gat.predict(C), I have >70% accuracy in the pre-stimulus area, and when I plot gat.predict(A), I have ~30%. This is off-diagonal accuracy - it's decoders around the N2 time window that generalize "backwards in time".
Basically, the pre-stimulus time window looks (as an example) like the N2 we get for B and C, but not for A.
(The data itself is baselined.)

This looks really silly.
It doesn't happen if I use gat.score because the prediction is as bad for A as it is good for C, only if I use predict (where the below-chance accuracy for A balances out the good results for C).

Is this simply inherent to the method (e.g. because the pre-stimulus brain process in our experiment simply happens to look like an N2), or am I doing something wrong?

kingjr, 2015-07-13:
I thought I had already answered that message. Have you solved your issue?

Else, and as it might be useful to other users:

* Electrophysiological components observed across time are not necessarily orthogonal with one another, so training at one time point, may lead to strong biases when you generalize it to another time point. You thus need to ensure to use 1. algorithms that output continuous predictions (regressors, probabilistic classifiers etc) and 2. use continuous scoring metrics (e.g. AUC)

* If you're not using balanced datasets (e.g. n_A = 100, n_B = 10), discrete metrics can be surprisingly bad for similar reasons than in the first point. Systematic biases can lead to ok mean performance but very poor individual sample predictions.

* You can use or build a scoring metrics that assesses accuracy per class, so as to check whether each class is predicted in a similar way.

If that doesn't help you, could you send me a downsampled example of your data?

PS: I tend to prefer high pass filtering than baselining, as the latter creates spurious increase of variance with increasing time, and can hide protocol errors that induce informative brain response before stim onset #truestory

jona-sassenhagen, 2015-08-04:
Okay I finally figured out something that works. Thanks @kingjr 

The solution to the baseline problem is to only ever look at subject average contrasts between the average baseline condition predictions (e.g,, the Y=0 condition in the training stage) and individual contrast predictions (e.g., everything else). That way, baseline effects cancel each other out.
(I'm very vary of using hard high pass filters to remove baseline effects for various reasons, although in this case it wouldn't have worked anyways because we don't have baseline *artefacts*, we have systematic pre-stimulus brain activity due to trial overlap.)

Solution looks something like this:

```
def fit_gat(epochs, conds=None):
    cond1, cond2 = conds
    scaler = StandardScaler()
    estimator = classifier(class_weight="auto")
    clf = Pipeline([('scaler', scaler), ('estimator', estimator)])
    gat = GAT(clf=clf, n_jobs=1)
    epochs = deepcopy(epochs)
    epochs.subtract_evoked()
    ep1, ep2 = epochs[cond1], epochs[cond2]
    ep1.events[:,2], ep2.events[:,2] = 0, 1
    ep1.event_id = {"_".join(cond1):0}
    ep2.event_id = {"_".join(cond2):1}
    eps = mne.epochs.concatenate_epochs((ep1, ep2))
    gat.fit(eps)
    gat.predict(eps)  # have to be extracted manually after training for now
    return gat

def predict_conds(things):
    epoch, gat = things
    """warning: overwrites old events"""
    epoch.events[:,2] = 1
    gat.predict_mode = "mean-prediction"
    return np.squeeze(np.asarray(gat.predict(epoch)))
```

closed 2015-08-04 by jona-sassenhagen

kingjr, 2015-08-04:
Great, although I'm not sure to have followed everything regarding the
basline...

Regarding your code:
- note that you can specify `y` a the fitting stage, so that you don't have
to overwrite `epochs.events` or do a deep copy.
- gat.predict returns y_pred so you could return this directly instead of
gat
- I think np.squeeze needs not asarray.

jona-sassenhagen, 2015-08-04:
Maybe I'll try and send you a description of what I mean wrt. the baseline thing. Could be an interesting case.