mne-tools/mne-python #2009: dipole plotting

mainakjas, 2015-04-23:
For dipoles changing over time, we want to see all the dipoles not just one.

cc @agramfort @yousrabk @Eric89GXL

agramfort, 2015-04-23:
this means our dipole plot obtained with dipole fit is not OK

Eric89GXL, 2015-04-23:
The `plot_dipole_locations` function will plot the first timepont for all dipoles in a list, `dipole.plot_locations()` will plot all the locations for one dipole. Do you want all the positions for all the dipoles in a list? That's a tough one because it would be hard to differentiate what points come from which dipoles at which times.

closed 2015-04-23 by agramfort

agramfort, 2015-04-23:
ok makes sense. I had missed this detail.

I pushed a fix in master.

closing this one.

thx