mne-tools/mne-python #2014: mne report scaling broken

mainakjas, 2015-04-23:
@teonlamont reported that scaling is broken. He gets reports with figures that have rotated figures like this one: https://plus.google.com/photos/albums/psjtfo8thlrdqa88m7qqn37ldkt235u54?pid=6140987061440567634&oid=108662407417175145691

@dengemann seems like git praise points to you :) https://github.com/mne-tools/mne-python/commit/011511b6846e5e76c8edf661f33e2bc35993c37c 

Also, I tried briefly and I got this: 

```py
RuntimeError: Could not set the fontsize
```
when `scale` was less than 1.0.

dengemann, 2015-04-24:
Could you please provide a minmal report examples to reproduce this?

I cannot:

https://gist.github.com/dengemann/e24941a27bc77056461d

mainakjas, 2015-04-24:
weird, I can't reproduce it any more too. I'll let @teonlamont post a script to reproduce the other bug while I try to figure out what I did to reproduce that.

mainakjas, 2015-04-24:
ok, this does it:

```py
from mne.report import Report
from mne.datasets import sample
from mne import read_evokeds
report = Report(verbose=True)

path = sample.data_path()
fname = path + '/MEG/sample/sample_audvis-ave.fif'
evoked = read_evokeds(fname, condition='Left Auditory', baseline=(None, 0),
					  verbose=True)
fig = evoked.plot(show=False)
fig.set_size_inches(3, 2)
report.add_figs_to_section(fig, captions='Left Auditory', section='evoked',
						   scale=0.9)
report.save('report.html', overwrite=True)
```

dengemann, 2015-04-24:
Well this is not really a bug ... you cannot scale this smaller than the smallest fontsize. It just does not throw a nice error.

dengemann, 2015-04-24:
see for scaling 0.5 with this example:

https://www.dropbox.com/s/oi7y927vk5jioif/Screenshot%202015-04-24%2003.24.14.png?dl=0

mainakjas, 2015-04-24:
okay, point taken. A nicer error would be good to have.

@teonlamont could you reproduce the other problem?

Eric89GXL, 2015-04-29:
If this is a release blocker it would be good to sort it out.

dengemann, 2015-04-29:
Teon, can you share a script to trigger the second issue? The scaling issue is not a release blocker.

teonlamont, 2015-04-29:
```python
p = epochs.plot_drop_log(show=False)
r.add_figs_to_section(p, 'Drop log of the events', 'Summary',
                      image_format='png', scale=.25)
```

Eric89GXL, 2015-04-29:
Can you add the necessary boilerplate before those lines using the sample dataset to make it cut-and-paste runnable? It is more DRY for you to do it instead of all of the rest of us who want to test it :)

teonlamont, 2015-04-29:
```python
import os.path as op
import mne
from mne.report import Report 
from mne.utils import _TempDir

r = Report()
tempdir = _TempDir()
data_path = op.join(mne.datasets.sample.data_path(), 'MEG', 'sample')
raw_file = op.join(data_path, 'sample_audvis_raw.fif')
event_file = op.join(data_path, 'sample_audvis_raw-eve.fif')

raw = mne.io.read_raw_fif(raw_file)
events = mne.read_events(event_file)
epochs = mne.Epochs(raw, events, None, tmin=-.1, tmax=.4,
                    reject={'grad': 1e-10}, preload=True)
p = epochs.plot_drop_log(show=False)
r.add_figs_to_section(p, 'Drop log of the events', 'Summary',
                      image_format='svg', scale=.25)
r.save(op.join(tempdir, 'test.html'))
```

Eric89GXL, 2015-04-29:
I actually see two bugs -- if the window is big, I see the one I assume you've reported:

[image]

But if I shrink the window, for some reason the top bar wraps and screws up the rendering further:

[image]

agramfort, 2015-04-29:
these are CSS issues we've always had right?

if it's the case it's not a blocker to me.

you can click on blue arrow in progress bar to go around the second bug.

dengemann, 2015-04-29:
Not a blocker, at these low values of scale I'm surprised it works at all.
And normally you should not need to downscale things. Rather upscale.

agramfort, 2015-04-29:
I agree.

mainakjas, 2015-04-29:
This script does not run for me and I get the error that I was getting earlier. Changing the scale to a larger value, I don't see inverted fonts. @teonlamont can you reproduce this error for larger scales? If not, we just need to add a more explicit error which checks against the current font size and warns the user if it's too low.

teonlamont, 2015-04-29:
ok. if it's easy to add a warning that would be nice, or even a note to say this only scales up images/figs.

Eric89GXL, 2015-04-29:
On it in #2039, see if you're happy @teonlamont @mainakjas

PR #2039 (2015-05-12): MRG FIX: Fix scaling
Closes #2014.

I'm pretty sure the docstring was wrong, too, so I changed it. At least it didn't do what it said it would do on my system (set the percentage). I'm not sure how you could have it both scale the figure by `scale` *and* have `scale` be the percentage of the page it should take up...
closed 2015-05-12 by dengemann