mne-tools/mne-python #2015: plot bad channel on top

mainakjas, 2015-04-24:
Currently, the bad channel is sandwiched among the other lines in the butterfly plot. It should be on top of the plot. This can be hopefully done by setting `zorder=-1`

cc @Eric89GXL

dengemann, 2015-04-24:
> Currently, the bad channel is sandwiched among the other lines in the butterfly plot. It should be on top of the plot. This can be hopefully done by setting zorder=-1

In think that should work when you draw 2 times. Once for the goods, once for the 'bads'.

mainakjas, 2015-04-24:
yes :)

Eric89GXL, 2015-04-24:
I think the bad channel should be plotted in the *back*, actually, which is what `zorder=-1` would do. `zorder=1` would move it to the front.

Eric89GXL, 2015-04-24:
...most of the time, you don't care about the bad channel. Although I guess if you don't want to see it you could do `exclude='bads'` so I guess you're right `zorder=1` should be done :)

dengemann, 2015-04-24:
@Eric89GXL the point here is to see the interpolated channels clearly, not to hid bads.

mainakjas, 2015-04-24:
actually, it would be cool to see the interpolated channels in a topo plot so that you can see that it actually follows the pattern of the neighbouring channels. But the topo plots currently don't visually mark the bad channels using a different color

dengemann, 2015-04-24:
@mainakjas it would look crappy if it was bad.

dengemann, 2015-04-24:
The topo would be dominated by extreme values or would show holes in cases where a flat channel remains.

PR #2073 (2015-05-10): ENH: Add butterfly channel picking
Closes #2071.
Closes #2015.

Ready for testing / review / merge @teonlamont.
closed 2015-05-10 by Eric89GXL