mne-tools/mne-python #2020: 'Opt' button doesn't do anything in plot_raw

choldgraf, 2015-04-24:
When I plot a Raw object with `plot`, there is a little box in the bottom left called `Opt`. Whenever I click it, nothing seems to happen. I looked into the code and it seems like this is supposed to bring up an options box of some kind. Does this work for other people? Maybe it's because I'm calling the function from an ipython notebook? Does it only work on certain graphics engines?

Eric89GXL, 2015-04-24:
It might only work for the Qt backend, not sure

choldgraf, 2015-04-24:
Hmm, I believe that I'm using Qt...

Eric89GXL, 2015-04-24:
you might need to be in "interactive mode", can you try from a standard Python terminal with `plt.ion()` called?

choldgraf, 2015-04-24:
Hmm, I've tried calling `plt.ion` both in the ipython notebook, as well as from an ipython session in the terminal. These are after calling `plt.switch_backend('Qt4Agg')`. If I put the code into a python script and run it, then a window pops up but immediately closes...

Eric89GXL, 2015-04-25:
I just ran this on OSX 64-bit Anaconda Python, and it worked fine:
```
>>> import matplotlib.pyplot as plt
>>> plt.ion()
>>> import mne
>>> data_path = mne.datasets.sample.data_path()
>>> raw = mne.io.Raw(data_path + '/MEG/sample/sample_audvis_raw.fif')
>>> raw.plot()
<matplotlib.figure.Figure object at 0x105f457d0>
>>>
```
Clicking "opt" brought up another window with a SSP selection. I can try it on Linux system Python on Monday, but I'm pretty sure it works there, too.

Eric89GXL, 2015-04-25:
Maybe the window that appears is falling behind the original one for you?

choldgraf, 2015-04-25:
Ah - do you need to have MEG information in there in order for that button
to work? I've just been creating Raw objects from random data using
RawArray.

Eric89GXL, 2015-04-25:
You might need projectors, yeah. It would be great to have a PR that greys out the button text if no projs are available!

Eric89GXL, 2015-04-25:
(FWIW projectors can be for MEG or EEG data, but using the `*Array` functions you'd probably end up with them only by using our preproc functions.)

choldgraf, 2015-04-25:
OK cool - at some point I'd like to improve the raw plotting anyway, so
maybe that's something to put on the to-do list. If there's ever an MNE
code sprint that I can attend, that'd be a fun thing to work on. I really
like how EEG-lab lets you change the scales of things on the fly, change
window size, etc.

Eric89GXL, 2015-04-25:
Yeah, I think there's an issue for that. The problem with waiting for an `mne` sprint is that @agramfort is in Europe so he tends to host them over there :)

choldgraf, 2015-04-25:
Fair enough... Maybe we can have a mne West sprint!

agramfort, 2015-04-25:
>
> Yeah, I think there's an issue for that. The problem with waiting for an
> mne sprint is that @agramfort <https://github.com/agramfort> is in Europe
> so he tends to host them over there :)
>
if you take my role i.e. write the grants to host them in the US I'd be
happy to join :)

dengemann, 2015-04-25:
@choldgraf the EEGLAB scaling stuff is btw. something we plan to implement in this year's google summer of code, also an improved epoch viewer.

choldgraf, 2015-04-26:
sounds good @dengemann, this would certainly be helpful. Perhaps I can do something to help when I am visiting in July!

choldgraf, 2015-04-26:
@Eric89GXL what kind of MNE userbase do you guys have up in Seattle? I wonder if there are pots of money at the UW / UCB institutes for Data Science for stuff like this...

Eric89GXL, 2015-04-27:
There are a number of people using it at ILABS, but I haven't really looked into getting money from the data science centers. If you have time to look, I'm happy to help out or talk to people locally.

choldgraf, 2015-04-27:
Cool - I will keep an eye open here at Berkeley. I know that the Berkeley
Institute for Data Science (BIDS) has a physical space w/ lots of breakout
rooms etc. It's pretty nice and they're pretty open to people utilizing it
for open source projects (e.g., the ipython team hangs out there pretty
frequently).

agramfort, 2015-04-27:
@Eric89GXL you should have a chat with @jakevdp at UW

@choldgraf you should do the same with @fperez at UCB

just doing data science match making on github... :)

Eric89GXL, 2015-04-27:
Actually -- @jakevdp or @fperez do you know of a good resource for hosting a ~2GB public dataset? We need to allow maintainers from multiple institutions (UW, Telecom ParisTech, MGH, etc.) to have ownership / modification permissions, and up-time is important. Do you have any experience with hosting services at UW or elsewhere (e.g., Amazon)? We would strongly prefer a free solution :)

choldgraf, 2015-04-27:
Didn't github just release a "Large File Storage" service [1]? I think 1GB
is free so this would be higher than that, but might be relevant.

Chris

[1] https://git-lfs.github.com/

choldgraf, 2015-04-29:
Random followup - I just chatted with some folks at BIDS here at UC Berkeley and they said that we could almost certainly get a room for a few days for a code sprint, possibly funding for some food etc as well but we'd need to look into that further.

@agramfort in your experience what kinds of resources are necessary in order to put on a code sprint? Perhaps having something over here on the west coast is more possible than thought.

agramfort, 2015-04-30:
if you fly people over you need travel money and housing.

FYI for the sprint this summer we asked for 15k€ and we hope to be able to
fly people from europe and a few from the US

choldgraf, 2015-04-30:
OK cool, that was my main question (about travel + lodging). I think something like BIDS can easily help with the lighter side of things, but stuff like lodging (unless that means staying on friends' couches) and stuff like travel (unless that means walking up and down the west coast ;) ) would take more to set up. 

Still might be possible, but maybe @Eric89GXL and I can just keep it in mind in the coming months. If flights across the Atlantic are an issue we could always do a simulcast or something, and conduct two sessions separately.

agramfort, 2015-04-30:
ok. Let's hope the french economy allows us to travel you guys over here :)

PR #2079 (2015-05-11): FIX: Fix projection in raw plotting
Closes #2020.

Also fixes a long-standing bug where projectors are not updated when marking bad channels in raw. Try the following code on `master`, then on this PR, scrolling down to `EEG001` and marking it bad:
```Python
import mne
raw = mne.io.Raw(mne.datasets.testing.data_path() +
                 '/MEG/sample/sample_audvis_trunc_raw.fif', preload=True)
raw._data[315, 300] = 1e-2
raw.plot()
```
I also modified the line width, it looks clearer to me -- see if you agree. In either case we should disable antialiasing because it doesn't do much for data samp [... cut]
closed 2015-05-11 by Eric89GXL