mne-tools/mne-python #2026: plotting bug with `plot_components` with EEG montage

teonlamont, 2015-04-26:
I am currently working with some biosemi data and I noticed one bug and one enhancement that should be addressed before the release.

the first is the bug: when plotting the IC for the EEG data, only one topo plot is currently being plotted while the rest are left empty.

[image]

the second is a feature enhancement: when using `ica.plot_components`, the plotting function should not default to `meg` especially when there are no meg channels present in the data. the default ch_type should be None. we should inspect the channel types, and if there are no meg channels, we should plot the EEG, if we have meg only, then plot meg, then if we have both, probably the default should be meg but that can be discussed.

dengemann, 2015-04-26:
bug fix yes, enhancement time is over. please noe new ENH requests.

dengemann, 2015-04-26:
@teonlamont I'm plotting ICs for EEG data almost every day. Can you share a snippet of code to reproduce this?

agramfort, 2015-04-26:
it's a backend pb I suspect. Try starting ipython with --gui=qt or
something along these lines

teonlamont, 2015-04-26:
how do you do that from a PC...? lol

dengemann, 2015-04-26:
@teonlamont what do you mean? On a PC ipython is supposed to work.

teonlamont, 2015-04-26:
with the flag --gui=qt. there's a separate application that is loaded for ipython

dengemann, 2015-04-26:
@teonlamont with anaconda there should be a linux terminal emulator.

teonlamont, 2015-04-26:
ok, @dengemann, I will make a separate issue for the `ch_type`. i think it would be a broader plotting feature to look at for any plot that requires a ch_type as input.

dengemann, 2015-04-26:
>  i think it would be a broader plotting feature to look at for any plot that requires a ch_type as input.

yes, I suggested that already a few times during seprate discussions. We should implement it once, carefully and consistently.

teonlamont, 2015-04-26:
@dengemann I will give it a look tomorrow and report back wrt to the linux terminal emulator. I just left the lab. I didn't know if it was our problem or a backend one.

teonlamont, 2015-04-26:
> yes, I suggested that already a few times during seprate discussions. We should implement it once, carefully and consistently.

yeah, I remembered it being discussed and i had thought it had been addressed.

dengemann, 2015-04-26:
open an issue and we'll re-discuss it.

PR #2036 (2015-04-29): FIX: Delay show to end
Closes #2026.

@teonlamont can you see if it fixes your problem? @dengemann the change from `show=show` to `show=False` on one line is more consistent with what we do elsewhere in the package, but I'm not sure if it messes with your use cases. Can you try the ICA plotting functions to see if you're happy with their behavior under this PR?
closed 2015-04-29 by agramfort