mne-tools/mne-python #2027: ENH ch_type = None should default to channel types presented

teonlamont, 2015-04-26:
when using ica.plot_components, the plotting function should not default to meg especially when there are no meg channels present in the data. the default ch_type should be None. we should inspect the channel types, and if there are no meg channels, we should plot the EEG, if we have meg only, then plot meg, then if we have both, probably the default should be meg but that can be discussed.

this should also be considered for other plotting functions that uses the `ch_type` argument.

dengemann, 2015-04-26:
My porposal would be:

- defalut to 'mag' if it has mags
- default to 'eeg' if does not have 'meg'

dengemann, 2015-04-26:
comment updated.

agramfort, 2015-04-27:
Sounds good

jona-sassenhagen, 2015-05-16:
This would be cool to have, I remember I was very confused by this a few times when I was starting out with MNE (as an EEG person).

teonlamont, 2015-05-16:
@jona-sassenhagen totally agree. I think it is because before MNE used to have a predominate MEG leaning since it's default filetype is FIF from Elekta. I think it is now coming around to be extremely powerful for EEG as well. This is why I would like to do away with blanket MEG defaults.

Eric89GXL, 2015-05-16:
I think @dengemann's proposal should work for this particular case, @teonlamont do you agree?

agramfort, 2015-05-16:
PR welcome

teonlamont, 2015-05-16:
@Eric89GXL i agree with the proposal

Eric89GXL, 2015-05-18:
I'm on this one

PR #2114 (2015-05-21): MRG ENH: Center and allow ch_type=None
Closes #2113.
Closes #2027.

Example code:
```
import os.path as op
import mne
from mne.datasets.testing import data_path
evoked_fname = op.join(data_path(), 'MEG', 'sample', 'sample_audvis-ave.fif')
evoked = mne.read_evokeds(evoked_fname)[0]
evoked.pick_types(meg=False, eeg=True)
evoked.plot_topomap()
```
On current `master` this throws an error. This is what we now get on this PR:

[image]

If change the last lines to select just the left EEG channels, with `ch_type='eeg'`:
```
left_ch = [ch['ch_name'] for ch in evoked.info['chs'] if ch['loc'][0] < 0]
e [... cut]
closed 2015-05-21 by dengemann