mne-tools/mne-python #2032: BUG: TypeError for scalings=None in mne.cov._estimate_rank_meeg_signals

adykstra, 2015-04-27:
when using <code>mne.cov._estimate_rank_meeg_signals</code> with scalings=None on the sample data, I get:
 
  File "/home/andrew/.local/lib/python2.7/site-packages/mne-0.9.dev0-py2.7.egg/mne/cov.py", line 1611, in _apply_scaling_array
    data *= scalings[:, np.newaxis]  # F - order

TypeError: 'NoneType' object has no attribute '__getitem__'

Yet, the doc says: "If None, no scaling will be applied."  

What should be the default behavior if scalings=None?  
    
    if scalings is None:
        scalings = dict(meg=1., grad=1., eeg=1.)

or

    if scalings is None:
        scalings = dict(mag=1e15, grad=1e13, eeg=1e6)

Also, I guess scalings should be defaulted to <code>None</code>?

agramfort, 2015-04-27:
can you share a gist to reproduce?

When scaling is None we tend to use our defaults.

adykstra, 2015-04-27:
https://gist.github.com/adykstra/b8ebfda2f153db17cc53

agramfort, 2015-04-27:
can this happen without calling directly the private function?

adykstra, 2015-04-27:
I'll check.

adykstra, 2015-04-27:
Yes ... 

    import mne
    from mne import io
    from mne.datasets import sample

    data_path = sample.data_path()
    raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
    event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'

    raw = io.Raw(raw_fname, preload=True)
    raw.estimate_rank(scalings=None)

agramfort, 2015-04-28:
thanks for checking and looking into it

PR #2064 (2015-05-06): MRG: Bugfix rank signals
Closes #2032.

Unifies our use of mutable defaults. I modified `_mutable_defaults` to only operate on a single parameter for simplicity. It also cleaned up the code quite a bit (as there were a bunch of `_mutable_defaults(('scalings', scalings))[0]` without it).
closed 2015-05-06 by Eric89GXL