mne-tools/mne-python #2034: winpython editable install

ImmanuelSamuel, 2015-04-28:
Installing mne in WinPython works but the editable install with "pip install -e +url" doesnt work.

agramfort, 2015-04-28:
for this to work you need git installed. if it's not the case please close.
Otherwise we have a bug to fix.

ImmanuelSamuel, 2015-04-28:
I have git installed ... I have a clone in the src folder in the winpython directory.

agramfort, 2015-04-29:
ok. What is the error you get?

ImmanuelSamuel, 2015-04-29:
I do not get any error. I just want to edit some of the functions in mne and then run my analysis which I am not able to do. If I run the function it runs everything except the lines I edited. It seems like an install in the non-editable mode.

agramfort, 2015-04-29:
ah. You're not doing it the right way. If you want edit the code you should
start with:

http://martinos.org/mne/dev/contributing.html#using-git

and in particular:

http://martinos.org/mne/dev/contributing.html#setting-up-the-fork-and-the-working-directory

hope this helps

ImmanuelSamuel, 2015-04-29:
I tried this but this installs the files in documents somewhere and is not contained within the winpython directory. Is there a way of editing setup.py to install within the winpython folder?

Eric89GXL, 2015-04-29:
When doing `python setup.py install`, a correctly configured Python setup will make it so that when you do `import x` later it will import the installed version. If it does not, it indicates that Python isn't configured correctly. When you run `python` does it actually execute the WinPython version of Python? Do you have another Python interpreter installed or configured? Have you sucessfully installed any other packages using `python setup.py install`?

ImmanuelSamuel, 2015-04-29:
Ok so I installed it using "python setup.py develop". I omitted the --user argument and it installed in the current directory but still I have the same problem. If I edit the script then it doesn't get updated. Is the --user argument necessary for it to work?

Eric89GXL, 2015-04-29:
Depends on your python setup. `--user` means install only for the current user, omitting it typically does it system-wide. If you have write permissions to the system-wide Python, it might not matter which one you should do. But `--user` should always work AFAIK, so you could try that one.

ImmanuelSamuel, 2015-04-29:
I am running spyder inside the winpython folder so I think it is using the same winpython version. If I uninstall then I cannot import it in spyder so i think it is the same version.

ImmanuelSamuel, 2015-04-29:
I just installed with --user but still same issue:

I run the following

chNos = list(range(1,32))
estr = 'E'
chanNames = [estr+str(chNo) for chNo in chNos]
info = mne.create_info(chanNames, 250.0, ch_types=None)

then change mne.create_info: delete everything in the function and replace with return 0.

If I run the function again I get an info object and not 0.

Eric89GXL, 2015-04-29:
Ahh right, you wanted an editable install. You'll need to use `--develop`, and possibly also `--user`. Then when you open Python, do:
```
import mne
mne
```
and it should print out where it imported MNE from. It's possible it's getting an older installed version, which you might need to delete for it to find the `--develop` version.

ImmanuelSamuel, 2015-04-29:
ok so I figured out what the problem is - I have to close spyder and repoen for the changes to reflect. I used import mne again without closing spyder but doesnt work. I used mne = reload(mne) which also doesnt update the function.

agramfort, 2015-04-29:
you discovered a standard behavior of the python virtual machine.

if you edit an imported file or module it will only be updated in a new
interpreter. it's for efficiency reason as otherwise import caching would
not work fine. Rather than closing spyder just open a new interpreter.

HTH

closed 2015-04-29 by agramfort

dengemann, 2015-04-29:
Checkout ipython %autoreload
Many google sources

ImmanuelSamuel, 2015-04-29:
yup just got that working. Thanks :)