mne-tools/mne-python #2035: Resample function optimizations

choldgraf, 2015-04-28:
Currently it takes me quite a long time to carry out a resample function. I've got about 120 channels with about 10 minutes of data, sampled at 5Khz, which equals about 120 x 3,000,000 data points. Obviously my first step in this process is to resample the data such that it's not so densely-sampled. However, this can take a really long time and use a lot of memory. 

I have two questions regarding this:

1. Is the resampling optimized at all in terms of making sure that the signal is a ^2 in length? In the past I've tried to pad the signal with 0's to the next pow2, but this can be a memory issue if there are enough datapoints. Another option is to find a close number with lots of factors, no?
1. What kind of speedup do you all see when you use CUDA? Is it worth the time to set up in terms of the performance boost?

Anyway just trying to see if anyone has thoughts on improving the speed of these functions. If more people use MNE that do ECoG research (and thus have little control over the recording parameters), it may prove useful.

agramfort, 2015-04-28:
the culprit is the resample method from scipy.signal that honestly sucks...
it does a dummy fft over the entire signal length... the problem has to be
fixed in scipy.

you might want to try: http://scikits.appspot.com/samplerate

Eric89GXL, 2015-04-28:
In my experince CUDA is about an order of magnitude faster. I'm pretty sure that was both for FIR filtering and resampling. What would really help the CPU case is using `upfirdn` as discussed in #1814.

choldgraf, 2015-04-28:
Yeah, the scipy signal processing is the one thing that makes me miss matlab (just a little bit). I've tried creating functions like `make_len_pow_2` that just dumbly the next power of 2 and appends 0's to the signal. That definitely speeds things up, but can explode the memory. What if something like that was added to the resample function as an option to `npad`? So if you put in a string like 'pow2' then it would find how many datapoints to add to make it a power of 2, then split it in half and add it to the front/back of the signal?

Eric89GXL, 2015-04-28:
+1 for the string idea, we already do something similar for filter lengths

choldgraf, 2015-04-29:
Did you already begin an upfirdn contribution to scipy? That may be outside of my signal processing chops :/

Eric89GXL, 2015-04-29:
No, and it's going to take some time to get right. So for now the padding to power of 2 should help.

Eric89GXL, 2015-04-29:
Someone actually has a SWIG'ed version available they gave me permission to relicense as BSD for scipy, but I'm not sure how complex that problem is going to be.

choldgraf, 2015-04-29:
Ah ok - for the resample string thing, do you know if there is a function
in python for "give me a number, and I will tell you the next highest
number with greater than N factors"? I've tried padding to a power of two,
but once you're in the higher regions it becomes memory-limited. That said,
this sounds like it may be too complicated to include natively :P.

Actually another thought might be to allow someone to specify the total
length that they want the signal to be, rather than the amount of padding.
E.g.: have some parameter `length`, take the diff from current length,
divide that in half, and add zeros to the front/back accordingly.

Eric89GXL, 2015-04-29:
No, I don't know of one. If memory is an issue, one thing that might fix that is iterating over channels instead of operating on them as a contiguous block (e.g., as `axis=1`). The overhead for a few hundred channels should be pretty negligible, but you'd want to test it to make sure.

choldgraf, 2015-04-29:
Ya that is true - if you do it for one channel at a time it's not a big
deal, so in that case you would basically do:

```python
  all_res_chans = []
  for chan in channels:
    chan = pad_to_power_2(chan)
    chan_res = resample(chan)
    all_res_chans.append(remove_padding(chan_res))
  make_back_into_Raw(all_res_chans)
```
or something like this

Eric89GXL, 2015-04-29:
Yeah, that's the pseudocode anyway. Getting it to work with the current resampling might take some work, but should be doable, at least for the `n_jobs=1` case. Sometimes we branch based on whether `n_jobs == 1` or not, and this would be a case where it could be useful to do so. You can't really avoid the memcpy-type operation with multiple jobs, unfortunately, so it will only work for a single-job case.

You're welcome to take a stab at it if you have time. I think if `n_jobs=1` you'd want to preallocate the new array, and fill it in row-by-row by resampling.

choldgraf, 2015-04-29:
mm that's a good idea. I will try to include this if I finish my stuff for
this week early. Have a plane flight tomorrow so maybe then.

Eric89GXL, 2015-10-21:
Hopefully I'll have time to actually make this PR work and get it into `scipy`:

https://github.com/scipy/scipy/pull/5186

Eric89GXL, 2015-12-18:
An update -- this should be made ~2x faster in the near term (months) by https://github.com/scipy/scipy/issues/5592 and the related https://github.com/scipy/scipy/pull/5610. By my estimate, using `upfirdn` through https://github.com/scipy/scipy/pull/5186 will require more time (~6-12 months).

choldgraf, 2015-12-21:
That's great - looking forward to not dreading my resampling steps :)

Eric89GXL, 2016-01-20:
FYI scipy/scipy#5610 (`upfirdn`) has been merged and I've started a `upfirdn`-based `resample` discussion in https://github.com/scipy/scipy/issues/5746.

choldgraf, 2016-01-20:
wohoo! wheels are a-turning :)

jona-sassenhagen, 2016-01-20:
Indeed, v. cook.

PR #2978 (2016-03-02): MRG: Faster raw resampling
This PR:

1. Adds `npad='auto'` mode that will set the pad to the shortest pad >= 100 samples that will result in a signal length that is a power of 2.

2. Deprecates the `npad=100` default to have `'auto'` become default in 0.12

3. Uses `rfft` and `irfft` for computations.

This means:

1. If there is an integer ratio of resampling rates (e.g., 1200 down to 100) then it should be *much* faster.

2. Memory usage should be reduced by a factor of 2 during computations.

Ready for review/merge from my end.

Closes #2035.
closed 2016-03-02 by Eric89GXL