mne-tools/mne-python #2040: preload_data is only for RawFIF

teonlamont, 2015-04-29:
I was just working with a RawEDF and I went to call `raw.preload_data`, but I noticed that this method didn't exist for this Raw class. It's currently only implemented for RawFIF. I also saw that there is a new (to me at least) parameter in `_read_segment` called `data_buffer` that is being used whereas the other modules don't have this. should we use this method's use?

agramfort, 2015-04-29:
as you can now load edf data without preloading yes we could add this
method. It means all raw classes should properly implement it. Clearly not
an issue for v0.9 but maybe a rainy sunday afternoon...

Eric89GXL, 2015-05-10:
@agramfort AFAIK only EDF and FIFF allow not preloading data. If that's the case then `preload_data` should automatically work for the other backends because they automatically set `self._preloaded = True` or whatever the property is, and `raw.preload_data()` checks that (or should check for it) and returns if data are already preloaded. So it should just be a matter of moving the method to `Base` and making sure it works with EDF.

agramfort, 2015-05-10:
can you give it a try?

Eric89GXL, 2015-05-10:
Yeah, maybe today or later this week

PR #2080 (2015-05-12): MRG ENH: Preload data
Closes #2040.

~~We'll see if Travis is happy. Still some optimizations to do, but the architecture is there.~~
closed 2015-05-12 by agramfort