mne-tools/mne-python #2044: Consistency in the use of Notes

teonlamont, 2015-04-29:
should Note at the beginning of some docs be treated differently than the Notes section in others?

ex.
```python
    def find_bads_ecg(self, inst, ch_name=None, threshold=None,
                      start=None, stop=None, l_freq=8, h_freq=16,
                      method='ctps', verbose=None):
        """Detect ECG related components using correlation

        Note. If no ECG channel is available, routine attempts to create
        an artificial ECG based on cross-channel averaging.

        Parameters
        ----------
        inst : instance of Raw, Epochs or Evoked
            Object to compute sources from.
```

vs.
```python
def read_events(filename, include=None, exclude=None, mask=0):
    """Reads events from fif or text file

    Parameters
    ----------
    filename : string
        Name of the input file.
        If the extension is .fif, events are read assuming
        the file is in FIF format, otherwise (e.g., .eve,
        .lst, .txt) events are read as coming from text.
        Note that new format event files do not contain
        the "time" column (used to be the second column).
    include : int | list | None
        A event id to include or a list of them.
        If None all events are included.
    exclude : int | list | None
        A event id to exclude or a list of them.
        If None no event is excluded. If include is not None
        the exclude parameter is ignored.
    mask : int
        The value of the digital mask to apply to the stim channel values.
        The default value is 0.

    Returns
    -------
    events: array, shape (n_events, 3)
        The list of events

    Notes
    -----
    This function will discard the offset line (i.e., first line with zero
    event number) if it is present in a text file.

    Working with downsampled data: Events that were computed before the data
    was decimated are no longer valid. Please recompute your events after
    decimation.
    """
```

lulopolar, 2015-04-29:
Not sure if this is relevant, but according to numpy docstring should be like this:

r"""A one-line summary that does not use variable names or the
function name.

Deprecation warning using Sphinx directives (optional):
.. note:: Deprecated in Package X.Y
    `ndobj_old` will be removed in Package X.Z, it is replaced by
    `ndobj_new` because the latter works also with array subclasses.

Several sentences providing an extended description of the code's
functionality.  Refer to variables using back-ticks, e.g. `var1`.

Parameters
----------
var : type
    Description of parameter `var`.  The type above can either
    refer to an actual Python type (e.g. ``int``), or describe the
    type of the variable in more detail, e.g. ``(N,) ndarray`` or
    ``array_like``.  Array_like means all those objects -- lists,
    nested lists, etc. -- that can be converted to an array.
x1, x2 : type
    Description of `x1`, `x2`.  Use this format when input
    parameters have exactly the same type, shape, and description.
long_var_name : {'hi', 'ho'}, optional
    Desciption of `long_var_name`.  When parameter can only
    assume one of a fixed set of values, those values can be listed
    in braces, with the default appearing first.
var_no_type
    Desciption of `var_no_type`.  The colon is ommited since the
    type is absent.
varopt : type, optional
    Description of `varopt` (Description of default values).

Returns
-------
type
    Explanation of anonymous return value of type ``type``.
outvar : type
    Explanation of return value named `outvar`.

Other Parameters (optional)
----------------
seldom_used_keywords : type
    Explanation

Raises (optional)
------
BadException
    Because you shouldn't have done that.

See Also (optional)
--------
otherfunc : relationship (optional)
newfunc : Relationship (optional), which could be fairly long, in
          which case the line wraps here.
obviousfunc (Use without description when the functionality is clear
            from the function name)

Notes (optional)
-----
Notes about the implementation algorithm (if needed). This can have
multiple paragraphs and include equations in LaTeX format.

Equations can be used inline like this:

    The value of :math:`\omega` is :math:`2 \pi f`.

Or in a separate line:

    .. math:: X(e^{j\omega } ) = x(n)e^{ - j\omega n}

For multiline equation write the LaTeX code below the math

    .. math::
        equation one \
        equation two

To include a citation use the text [1]_ or the relevant number.

References (optional)
----------
.. [1] Doe, J. K., et al. (20XX). Paper title. Abbreviated Journal
   name, Vol(Issue), pp-pp.

Examples (optional)
--------
These are written in doctest format, and should illustrate how to
use the function.  This section is strongly encouraged but optional.
Separate multiple examples with a comment leaving a blank line above
and below it.

This is my first example:

>>> a = [1,2,3]
>>> print [x + 3 for x in a]
[4, 5, 6]

Comment explaining second examples.

>>> np.add([1, 2], [3, 4])
array([4, 6])

"""

lulopolar, 2015-04-29:
Also, I wrote a little code snippet that implements a form to dynamically add the docstring with the numpy template. It works on textmate and sublimetext. I’d be happy to share it if anyone is interested.

agramfort, 2015-04-29:
I am interested. Welcome Camilo ;)

lulopolar, 2015-04-29:
Hi Alex! here they are: https://gist.github.com/lulopolar
- Camilo

Eric89GXL, 2015-05-10:
@teonlamont my feeling is that really import notes (like about caveats and such) should go at the top so that people are less likely to miss them, whereas details like the implementation algorithm should go in the `Notes` section. I think this is how we currently have it organized. If people agree, we can close this after mentioning that we should clarify this in the FAQ/doc issue that is currently open.

closed 2015-07-24 by teonlamont