mne-tools/mne-python #2047: Report with ica.plot_components

teonlamont, 2015-04-30:
If ica yields more than 20 components, it creates two figures. the caption and the comment arguments will crash if you're unaware and only assign a single string.

agramfort, 2015-04-30:
maybe a good easy fix for @jaeilepp ?

Eric89GXL, 2015-05-13:
It looks like this is our last blocker. @teonlamont @dengemann @mainakjas do one of you have time to fix this?

mainakjas, 2015-05-13:
@teonlamont can you post a script to reproduce just so that I'm sure it's not a usage issue

dengemann, 2015-05-13:
Not tonight. Maybe very late.

teonlamont, 2015-05-14:
```python
from mne.report import Report

r = Report()
path = mne.datasets.sample.data_path()
raw_file = path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_file, preload=True)

ica = mne.preprocessing.ICA(.9, method='infomax')
ica.fit(raw, decim=4)

p = ica.plot_components(show=False)
r.add_figs_to_section(p, captions='ICs', section='IC')
```

jaeilepp, 2015-05-14:
To me, this does look like a usage issue. It is up to the user to make sure every figure has a valid caption. If the user gives a single string, it would be easy to multiply it to every figure to satisfy the validator in the code, but with a list of strings the logic wouldn't work.

Eric89GXL, 2015-05-14:
So maybe just a clear error like `2 figures were passed but only 1 caption was provided` would work?

dengemann, 2015-05-14:
Sounds good to me.

Eric89GXL, 2015-05-14:
any volunteer?

jaeilepp, 2015-05-14:
I can do it. At the moment it reports 'Captions and report items must have the same length.'. I'll just add the number of items to the message...

Eric89GXL, 2015-05-14:
@teonlamont okay with you?

teonlamont, 2015-05-14:
well, there should be an option to make all the ICs one fig instead of generating multiple figures. If I am batching subjects and one subject has 20 (good) and another has 21 (fail). this is because we are automatically limiting the number of IC per figure.

dengemann, 2015-05-14:
> well, there should be an option to make all the ICs one fig instead of generating multiple figures. If I am batching subjects and one subject has 20 (good) and another has 21 (fail). this is because we are automatically limiting the number of IC per figure.

No more features for now.
Handle types? If it's a list, pass list of names?

teonlamont, 2015-05-14:
the thing is that you don't know if it is going to exceed the IC figure limit or not. are you suggesting that the user should have in their code an isinstance check whether it is a figure or a list of figures?

dengemann, 2015-05-14:
No you in that case. Please improve stuff on 0.10 master. We need to be serious about feature stops. This is no bug. The plot does what the docs say.

jaeilepp, 2015-05-14:
It does seem weird to me too that there is a constant (20) in the code that determines the number of figures, but I'm not sure what is the best way for visualizing ICA components. Maybe we'll discuss it over the GSOC.

teonlamont, 2015-05-14:
the simplest solution without really adding a feature would be to have a parameter that defaults to this. it's not in the documentation anywhere that this is the behavior. there is only one inline note that makes this clear. it's just frustrating when I am trying to make a simple report of ICs it breaks because there's this arbitrary IC limit that isn't explained anywhere.

dengemann, 2015-05-14:
>  it's not in the documentation anywhere that this is the behavior. there is only one inline note that makes this clear

```
Signature: mne.viz.plot_ica_components(ica, picks=None, ch_type='mag', res=64, layout=None, vmin=None, vmax=None, cmap='RdBu_r', sensors=True, colorbar=False, title=None, show=True, outlines='head', contours=6, image_interp='bilinear')olated sensor topogrpahy.

Parameters
----------
ica : instance of mne.preprocessing.ICA
    The ICA solution.
picks : int | array-like | None
    The indices of the sources to be plotted.
    If None all are plotted in batches of 20.
...

Returns
-------
fig : instance of matplotlib.pyplot.Figure or list
    The figure object(s).
```

teonlamont, 2015-05-14:
ok, didn't see it there.

-teon

dengemann, 2015-05-14:
@teonlamont no question there is room for improvements. Let's be sharp about addressing bugs for now.

teonlamont, 2015-05-14:
ok fine, I personally don't think this is how the plotting should be
handled but I don't want this to be the issue to hold up the release.

-teon

dengemann, 2015-05-14:
We agree. But development is over for 0.9. Next cycle.

agramfort, 2015-05-14:
to sum up what's the plan here?

dengemann, 2015-05-14:
MNE 0.10, no bug. ENH.

teonlamont, 2015-05-14:
will we add a parameter for the n_ics per figure? Default is k n_ics, and None plots all ics in one figure?

dengemann, 2015-05-14:
> will we add a parameter for the n_ics per figure?

ok with me.

> Default is k n_ics

20 :)

> None plots all ics in one figure?

will look rather crappy...

If you can make it nice go for it. I could not.

Eric89GXL, 2015-11-25:
@teonlamont you still interested in having this? It's been a few months since last interest.

teonlamont, 2015-12-13:
maybe @jaeilepp, if you have time to look into this for a simple solution, no rush. if not, I will look into this at some point in the future.

Eric89GXL, 2016-03-16:
This one keeps getting bumped from version to version so I'm going to remove the milestone marker

teonbrooks, 2018-08-24:
closing. just thought of an easy fix to my complaint using https://github.com/mne-tools/mne-python/issues/2047#issuecomment-102028134 above:

from mne.report import Report

```python
r = Report()
path = mne.datasets.sample.data_path()
raw_file = path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_file, preload=True)

ica = mne.preprocessing.ICA(.9, method='infomax')
ica.fit(raw, decim=4)

p = ica.plot_components(show=False)
n_captions = len(p)
captions=['ICs %d of %d' %(ii+1, len(p)) for ii in range(len(p))]
r.add_figs_to_section(p, captions=captions, section='IC')
```

closed 2018-08-24 by teonbrooks