mne-tools/mne-python #2048: epochs.py raises error with argument proj=False

dbdq, 2015-04-30:
Hello,

On epochs.py line 682, inside class Epochs:
proj = proj or raw.proj  # proj is on when applied in Raw

Here, proj will become None if raw.proj==None.
It's because in Python, (None or False) == False, whereas (False or None) == None.
I believe it's confusing anyway to compare a None type value with a boolean value.

Sometimes, I have raw arrays having raw.proj==None, and it crashes the program by raising an error after the following line:
if proj not in [True, 'delayed', False]:

So for my code, I just added the following line to fix:
if proj == None: proj= False

after each line of "proj = proj or raw.proj". There were only 2 such lines affected.

agramfort, 2015-04-30:
can you share full code snippet to reproduce the error?

is it still failing on current master?

thx

dbdq, 2015-04-30:
Hello, after re-installing, I could confirm that this error does not happen in the master.
I think I was using an old version (0.8.x). Sorry for that.

Eric89GXL, 2015-04-30:
No problem, thanks for checking! Let us know if you hit any other problems.

closed 2015-04-30 by Eric89GXL