mne-tools/mne-python #2050: EHN: report save when error?

kingjr, 2015-05-01:
@dengemann Do you know whether it would be feasible to automatically save the report if there's an error and catch the latter without having to explicitly put the report.save() a try/except condition?

dengemann, 2015-05-01:
Sounds like that would take some hacking ... for example via with statements. I'm not sure we want to untroduce a syntax like that:

```Python

with Report() as report:
    # do stufff

```

dengemann, 2015-05-01:
Leaving the with block-- whether due to a crash or whether due to success-- the report would then be saved for eaxmple ...

Eric89GXL, 2015-05-01:
that looks like a decent syntax

are the errors in the report code or your local code?

dengemann, 2015-05-01:
> that looks like a decent syntax

just a bit more elegant than wrigin try/catch blocks in your script.

kingjr, 2015-05-01:
these are for my error, but yes I'm indeed trying to find a way to avoid doing
```
try:
   launch()
   a()
   very()
   long()
   pipeline()
report.save()
```
in all my scripts

dengemann, 2015-05-01:
You would the write instead:

```
with Report as report:
   launch()
   a()
   very()
   long()
   pipeline()
```

dengemann, 2015-05-01:
1 line saved ...

kingjr, 2015-05-01:
Ok, I need to think more. Maybe there's a better way of setting the report than what we're doing in the current pipeline form.

kingjr, 2015-05-01:
How about having an automatic saving + overwrite everytime the report is modified?
```
report, run_id, results_dir, logger = setup_provenance(script=__file__, results_dir=results_dir, continuous_saving=True)
```

dengemann, 2015-05-01:
Maybe let's move this discussion to meeg-preprocessing. We could subclass report inside and overwrite the save method to allow fo such things if requested.

kingjr, 2015-05-01:
ok

closed 2015-05-01 by kingjr

dengemann, 2015-05-01:
To make this work at the MNE-Python level we would need to find a scalable
way of updatting HTML +  dumping to files incrementally. The brute way
(overwriting) might be less performant.

kingjr, 2015-05-01:
It's less performant, but most HTML are not that heavy, so overwriting wouldn't be a major issue, should it? (But yes, ideally, it would be incremental,  but IMO that'd be a big piece of work, for a small feature)

dengemann, 2015-05-01:
let's check it out in our meeg-preprocessing playground sandbox