mne-tools/mne-python #2051: Initialise coil types

ImmanuelSamuel, 2015-05-02:
How to initialize coil types in info object?

```
chans = mne.channels.read_montage('BrainAmpMRIPlus_32sfp', ch_names=None, path= datapath, unit='mm', transform=False)

chanNames = chans.ch_names
info = mne.create_info(chanNames, 250.0, ch_types='eeg')
layout = mne.channels.find_layout(info, ch_type='eeg', exclude=None)

eventOnset = np.array(range(1,41))*2
eventZero = np.ones(40)
eventId = np.ones(40)
events = np.transpose(np.vstack((eventOnset, eventZero, eventId)))

epochs = mne.EpochsArray(cueLtemp, info, events, tmin=0, event_id=None, reject=None, flat=None, reject_tmin=None, reject_tmax=None, verbose=None)

epochs.set_montage(chans)

epochs.plot_projs_topomap(ch_type='eeg', layout=layout)
```

I get RuntimeError: No EEG channels present. Cannot find EEG layout. when I run mne.channels.find_layout()
If I use create_info with ch_types = 'eeg' shouldn't the coil type be eeg also? what exactly is the coil type and coil trans anyway? Should they be initialised

Eric89GXL, 2015-05-02:
Can you turn this into a minimal example? That would mean sharing your Montage file, or making an example that fails with some sample verison, and creating all variables from real data or numpy (e.g., the cuLtemp file is not defined above). That will allow us to test locally.

Is this on latest master?

@teonlamont you have time to tackle this?

ImmanuelSamuel, 2015-05-02:
Try running the following
```
chNos = list(range(1,32))
estr = 'E'
chanNames = [estr+str(chNo) for chNo in chNos]
info = mne.create_info(chanNames, 250.0, ch_types='eeg')
layout = mne.channels.find_layout(info, ch_type='eeg', exclude=None)
```

Eric89GXL, 2015-05-02:
Ahh, thanks for the more minimal example, I do have the same failure. That should help find the cause.

ImmanuelSamuel, 2015-05-02:
Montage file in .sfp format:
```
     FP1	 -26.133	  80.784	 -4.0011
     FP2	  26.133	  80.784	 -4.0011
      F3	   -48.2	  57.551	   39.87
      F4	  48.143	  57.584	  39.892
      C3	 -63.171	       0	  56.872
      C4	  63.167	       0	  56.876
      P3	   -48.2	 -57.551	   39.87
      P4	  48.143	 -57.584	  39.892
      O1	 -26.133	 -80.784	 -4.0011
      O2	  26.133	 -80.784	 -4.0011
      F7	 -68.423	  49.871	 -7.4895
      F8	  68.384	  49.927	 -7.4851
      T7	 -84.539	       0	 -8.8451
      T8	  84.539	       0	 -8.8451
      P7	 -68.423	 -49.871	 -7.4895
      P8	  68.384	 -49.927	 -7.4851
      Fz	       0	  60.738	  59.463
      Cz	       0	       0	      85
      Pz	       0	 -60.738	  59.463
      Oz	       0	 -84.981	  -1.786
     FC1	 -32.351	  32.436	  71.598
     FC2	  32.351	  32.436	  71.598
     CP1	 -32.351	 -32.436	  71.598
     CP2	  32.351	 -32.436	  71.598
     FC5	 -76.247	  28.763	  24.167
     FC6	  76.247	  28.763	  24.167
     CP5	 -76.247	 -28.763	  24.167
     CP6	  76.247	 -28.763	  24.167
     TP9	 -70.076	 -23.302	 -42.088
    TP10	  70.076	 -23.302	 -42.088
     POz	       0	 -79.026	  31.304
```

and CueLtemp is

```
data = numpy.random.rand(40,31,500)
```

ImmanuelSamuel, 2015-05-02:
```
chNos = list(range(1,32))
estr = 'E'
chanNames = [estr+str(chNo) for chNo in chNos]

info = mne.create_info(chanNames, 250.0, ch_types='eeg')
layout = mne.channels.find_layout(info, ch_type='eeg', exclude=None)
```

ImmanuelSamuel, 2015-05-02:
what happens is that in the info.chs there is a coil type which is not initialised to eeg even though in create info I type ch_type = 'eeg'

Eric89GXL, 2015-05-02:
yeah, I see it, fixing now...

Eric89GXL, 2015-05-02:
you should have `exclude=[]` in your script BTW, not `exclude=None`

Eric89GXL, 2015-05-02:
See #2053, can you try my branch to see if it works?

ImmanuelSamuel, 2015-05-02:
New error

```
RuntimeError: No digitization points found.
```

Eric89GXL, 2015-05-02:
Okay, I'll let @teonlamont take over then, I think he's worked mostly on the Montage stuff. You're also welcome to take a look at the code if you have time.

ImmanuelSamuel, 2015-05-02:
Cool ... I am just starting to enter python github and other stuff. Will try to contribute to mne. Thank you.

agramfort, 2015-05-03:
ok we've made progress. You need to use the new DigMontage class in order
to specify EEG channels locations in 3d.

teonlamont, 2015-05-03:
you should only call find_layout after you have set the montage to your inst. find_layout relies on using either coil type info or the channel positions and those are not specified when the montage is set.

teonlamont, 2015-05-03:
@ImmanuelSamuel try this:
```python
epochs = mne.EpochsArray(cueLtemp, info, events, tmin=0, event_id=None, reject=None, flat=None, reject_tmin=None, reject_tmax=None, verbose=None)

epochs.set_montage(chans)
epochs.plot_projs_topomap()
```

or alternatively
```python
epochs = mne.EpochsArray(cueLtemp, info, events, tmin=0, event_id=None, reject=None, flat=None, reject_tmin=None, reject_tmax=None, verbose=None)

epochs.set_montage(chans)
layout = mne.channels.find_layout(epochs.info, ch_type='eeg', exclude=None)

epochs.plot_projs_topomap(ch_type='eeg', layout=layout)
```

you should make sure you set you've added your projs to the epochs.

teonlamont, 2015-05-03:
@Eric89GXL @agramfort do you think we should have montage as an attribute of the inst?
what it would do is make Montage object that contains the channel position, names, and a plotting features. it is useful for inspection of the channel info.

agramfort, 2015-05-03:
adding a montage param to create_info maybe. But it'd love to have only 1
way to do things. It's then way easier to document.

teonlamont, 2015-05-03:
+1 for adding montage param to create_info

dengemann, 2015-05-03:
>  But it'd love to have only 1
way to do things. It's then way easier to document.

+1

>  do you think we should have montage as an attribute of the inst?
what it would do is make Montage object that contains the channel position, names, and a plotting features. 

-1

This is not neede for MEG for example and we already have too many attributes.

teonlamont, 2015-05-03:
montage argument has been added to the create_info: #2054

teonlamont, 2015-05-03:
if no one else is doing it, I will try to write up some things to go along with the montage classes.

PR #2053 (2015-05-03): FIX: Fix channel type
Closes #2051.

@ImmanuelSamuel can you see if this fixes your problem?
closed 2015-05-03 by Eric89GXL