mne-tools/mne-python #2052: save video from mne.morph_data

atalanti, 2015-05-02:
Dear mne-python users,

I have solved the source estimation problem from eeg data and using morph_data with time viewer parameter have managed to visualize the left hemisphere. Using the time gui I can select a single time among a range ( this range is the one that my epoch is defined) and view the source space that specific time. Is there any way to record a video within the whole time period range? 
If not, how can I save the visualization of the single time instance?
Thank you in advance,
atalanti

Eric89GXL, 2015-05-02:
Hey Atalanti,

mne-python uses a PySurfer Brain object to do surface source estimate
visualizations. Recently movie exporting functionality has been added to
PySurfer, can you look into that and see if it will do what you want?

https://github.com/nipy/PySurfer/blob/master/examples/save_movie.py

You should also be able to use brain.save_image (or something similar) to
save the currently displayed image.

Cheers,
Eric

atalanti, 2015-05-03:
Hi Eric,
Thank you for your reply.
I tried what you told me using the example that you sent me. However, I get this error:
AttributeError: 'Brain' object has no attribute 'save_movie'

atalanti

Eric89GXL, 2015-05-03:
Are you on latest master of PySurfer? It's probably not in a release yet.

atalanti, 2015-05-03:
Yes, I checked it. I am on the latest master of Pysurfer. I also reed the code at the viz.py (into the folder surfer) where the class brain is defined, and there is a def "save_movie".  I do not understand why I get this error then.

Eric89GXL, 2015-05-04:
If you are looking at the code of `viz.py` and see functions that aren't available when you actually use PySurfer, it indicates that the version of PySurfer you import from Python is not the one that you are looking at the code for. In your mne-python example, try doing this:
```
>>> import surfer
>>> surfer
```
It should print out the path to the PySurfer that is actually being imported. That version is probably out of date, and you should remove it.

Eric89GXL, 2015-05-05:
Great, thanks @atalanti

closed 2015-05-05 by Eric89GXL