mne-tools/mne-python #2056: log underflow in mne.stats.linear_regression

TalLinzen, 2015-05-03:
_fit_lm returns a quantity called mlog10_p_val for easier plotting of p-values. this fails when the p-value is 0.

https://github.com/mne-tools/mne-python/blob/master/mne/stats/regression.py#L134

one possible solution would to have some reasonable minimal p-value, e.g. instead of
```python
mlog10_p_val[predictor] = -np.log10(p_val[predictor])
'''

you could do something like

```python
mlog10_p_val[predictor] = -np.log10(np.clip(p_val[predictor]), 0.0000000001, 1))
'''

dengemann, 2015-05-04:
This is a dupe. See #1917.  Yeah, I think we should do something along the lines proposed by you, maybe using ```np.finfo(np.float64).tiny
```

agramfort, 2015-05-04:
+1 for using np.finfo. min pvalue should be around 1e-14

dengemann, 2015-05-04:
@TalLinzen PR welcome :)

TalLinzen, 2015-05-04:
sure, though if the min pvalue is around 1e-14 and `np.finfo(np.float64).tiny` is 2e-308 (on my computer) then it might be better to go for 1e-14, no?

TalLinzen, 2015-05-04:
hmm on a second thought I'm not sure what you meant when you said that the min pvalue is 1e-14. ;)

agramfort, 2015-05-04:
try:

In [1]: np.finfo(np.float64).eps
Out[1]: 2.2204460492503131e-16

Eric89GXL, 2015-05-04:
`tiny` probably has to do with the smallest double precision floating point value before going denormalized, whereas `eps` should be the smallest `d` for which `1 + d > 1`. `eps` probably makes more sense as a meaningful value in this context, but it could actually prove to be too large since eventually we deal with the `log10` of the value -- that will depend on the specific calculations being done to calculate `p`. I guess we could start out with `eps` and if we decide it's clipping too many values, change it to `tiny`. Hopefully it won't matter which one we use either way.

teonlamont, 2015-06-22:
is there a solution for this? I just got the same RuntimeWarning

PR #2713 (2015-12-17): MRG: Fix regression
Closes #2056.
Closes #1917.

Gets rid of warnings about divide by zero and underflows in linear regression by protecting against `stderr == 0`. It effectively forces all values in mlog10_p_val to be in the range `[-9.64e-17, 0]` (i.e., `[np.log10(1. - np.finfo(np.float64).eps), 0]`).

@dengemann I'd like to also take care of #1917 with this, but practically I'm not sure if we should make any changes to the code -- it might be doing the right thing already. I think if you get a p-value of 1., you should get a `0` in your `mlog10_p_val` and `np.inf` in your `t_val`, and deal  [... cut]
closed 2015-12-17 by dengemann