mne-tools/mne-python #2057: Reference for the cv-based shrunk covariance estimation

lulopolar, 2015-05-04:
Hi! 
This might be an appropriate reference for the shrunk covariance estimation method:
Warton, D. I. (2008). Penalized normal likelihood and ridge regularization of correlation and covariance matrices. Journal of the American Statistical Association. http://doi.org/10.1198/016214508000000021
Might be worth while including it in the covariance documentation

agramfort, 2015-05-04:
fine with me. It's however a problem that I cannot access this website :( :(

dengemann, 2015-05-04:
@lulopolar thanks for sharing it. Indeed the CV version of the shrunk covariance is without any ref. But basically the principal ref is the Ledoit-Wolf we cite in the paper and here. The only difference is, that you tune the alpha parameter, but structurally it is the same... A few implementation details are in fact not so well documented, e.g. the fact that each grid search is performed separately for sensor types.

lulopolar, 2015-05-04:
@agramfort: The paper freely available via google search :)
@dengemann: Thanks for the explanation! What I thought was neat about the JASA paper was that it show how cv specifically works in this covariance estimation problem, as well as the properties of the the estimator. It makes a lot of sense to have one regularization parameter per sensor type. In the JASA paper, they suggest to penalize the sample correlation instead. The difference might be then that the off-diagonal blocks of the correlation matrix corresponding to different sensor types might be non-zero, where as in the covariance are effectively zero.

closed 2015-05-06 by lulopolar