mne-tools/mne-python #2060: Converting .set -> .edf -> .fif, stimulus codes lost (from mne_analysis mailing list)

johnmarktaylor, 2015-05-05:
Hello,

I was wondering if someone could help me with a problem I've been having. We are trying to apply the pipeline described in the Cookbook to some EEG data that was collected in the .cnt format (Neuroscan). To convert this data to the .fif format, we are first converting it to an EEGLab .set file, which is converted to a .edf, which we are trying to convert to a .fif file. All of the data makes it through alright, with the exception of the stimulus codes. The stimulus codes do make it to the .edf stage, where they are represented in the channel labeled "status" as skinny spikes whose height corresponds to the value of the stimulus codes (in .cnt and .set formats, they are represented as numerical codes from 0 to 255). Here's the problem: when we then try to convert the .edf files to the .fif format using mne_edf2fif, the stimulus codes seem to get lost in transit (when we try to view the .fif file using mne_browse_raw, it doesn't find any events).

Does anyone know the fix for this? We've tried adding the option "--digtrig status" to mne_browse_raw so it knows where to look for the codes, but this doesn't seem to help anything. I noticed the "tal" and "annotmap" options in mne_ed2fiff, but these seem to apply to edf+ files, not edf files. I have no experience with the edf or fif formats, so there's a possibility I'm just misunderstanding something.

Thanks in advance for any help,

JohnMark Taylor
Yale Human Neuroscience Lab

Eric89GXL, 2015-05-05:
Relevant files:

https://www.dropbox.com/sh/fxvzjuwe0uh2qds/AADAL6_rBDlTE3c9gEMkPakLa?dl=0

The .set -> EDF converter converts the codes. Our EDF reader converts them to decimals because it divides be 1e6, then casts to int so everything becomes zero, i.e. gains here is `1e-6` for every channel including stim:

https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L183

So when it gets here it is cast to zeros:

https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L273

@teonlamont WDYT?

@johnmarktaylor in the meantime, one thing you can do is load your FIF file in mne-python, and scale it back up yourself. If you do this:
```
>>> import mne
>>> raw = mne.io.Raw(fname, preload=True)
>>> raw[-1, :][0]
```
you will see the values of your stimulus channel. As a workaround, you can scale them up by whatever factor you like, this will probably work:
```
>>> import numpy as np
>>> import mne
>>> raw = mne.io.Raw('tb9226_postlearning_withquestions.fif', preload=True)
>>> raw._data[-1, :] = np.round(raw._data[-1, :] * 1e6)
>>> mne.find_events(raw, 'Status')
Trigger channel contains negative values. Taking absolute value.
array([[ 20972,      0,    201],
       [ 21836,      0,     40],
       [ 22712,      0,    101],
...
>>> raw.save('tb9226_postlearneng_withquestions_fixed_raw.fif')
```
Then you should be able to load it with `mne_browse_raw --digtrig Status` (or just use `raw.plot()` and see your events.

johnmarktaylor, 2015-05-05:
Hi Eric, 

Just applied that fix and now it works perfectly! Thanks so much for your amazing prompt help on this, you're a lifesaver. 

Cheers,

JohnMark

agramfort, 2015-05-05:
unless someone volunteer to write a native python reader for cnt files I am closing.

glad it worked !

closed 2015-05-05 by agramfort

Eric89GXL, 2015-05-05:
Or there could be some fix in the EDF reader, like `stim_scaling=1.` where they could use `1e6` and it would work. But we can wait to see if others hit the issue, too.

teonlamont, 2015-05-05:
strange, the header info in this edf was written with scaling on the stim channel: https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L413. I guess this is a symptom of the set -> edf converter. This isn't a problem with native edf formats.