mne-tools/mne-python #2062: FIX: whitening example is broken afer show parameter

dengemann, 2015-05-05:
You should build docs from time to time dudes :)

dengemann, 2015-05-05:
cc @mainakjas @teonlamont

dengemann, 2015-05-05:
sorry gentlemen, issue on my end.

closed 2015-05-05 by dengemann

mainakjas, 2015-05-05:
:) okay, but both of us are unable to build docs due to some dependency issue

Eric89GXL, 2015-05-06:
Can you try now? I changed an exception and at least the noplot might work
now, assuming your error was on mayavi import.

dengemann, 2015-05-06:
Closed, deems fine.

Eric89GXL, 2015-05-06:
Sorry, I was talking to Mainak about fixing the doc building problem not
this issue

dengemann, 2015-05-06:
ok.

mainakjas, 2015-05-06:
No, it's the same error as before. This is with `make html-noplot`

```py
Traceback (most recent call last):
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/cmdline.py", line 245, in main
    app.build(opts.force_all, filenames)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/application.py", line 264, in build
    self.builder.build_update()
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/builders/__init__.py", line 245, in build_update
    'out of date' % len(to_build))
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/builders/__init__.py", line 316, in build
    self.write(docnames, list(updated_docnames), method)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/builders/__init__.py", line 354, in write
    self._write_serial(sorted(docnames), warnings)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/builders/__init__.py", line 362, in _write_serial
    self.write_doc(docname, doctree)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/builders/html.py", line 439, in write_doc
    self.docwriter.write(doctree, destination)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/docutils/writers/__init__.py", line 80, in write
    self.translate()
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/writers/html.py", line 53, in translate
    self.document.walkabout(visitor)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/docutils/nodes.py", line 174, in walkabout
    if child.walkabout(visitor):
  File "/home/mainak/anaconda/lib/python2.7/site-packages/docutils/nodes.py", line 174, in walkabout
    if child.walkabout(visitor):
  File "/home/mainak/anaconda/lib/python2.7/site-packages/docutils/nodes.py", line 174, in walkabout
    if child.walkabout(visitor):
  File "/home/mainak/anaconda/lib/python2.7/site-packages/docutils/nodes.py", line 174, in walkabout
    if child.walkabout(visitor):
  File "/home/mainak/anaconda/lib/python2.7/site-packages/docutils/nodes.py", line 166, in walkabout
    visitor.dispatch_visit(self)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/docutils/nodes.py", line 1882, in dispatch_visit
    return method(node)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/writers/html.py", line 720, in visit_literal_block
    HTMLTranslator.visit_literal_block(self, node)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/writers/html.py", line 331, in visit_literal_block
    **highlight_args)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Sphinx-1.3.1-py2.7.egg/sphinx/highlighting.py", line 185, in highlight_block
    hlsource = highlight(source, lexer, formatter)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/__init__.py", line 87, in highlight
    return format(lex(code, lexer), formatter, outfile)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/__init__.py", line 66, in format
    formatter.format(tokens, realoutfile)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/formatter.py", line 95, in format
    return self.format_unencoded(tokensource, outfile)
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/formatters/html.py", line 838, in format_unencoded
    for t, piece in source:
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/formatters/html.py", line 684, in _wrap_div
    for tup in inner:
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/formatters/html.py", line 697, in _wrap_pre
    for tup in inner:
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/formatters/html.py", line 716, in _format_lines
    for ttype, value in tokensource:
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/filter.py", line 20, in _apply
    for token in filter_.filter(lexer, stream):
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/filters/__init__.py", line 194, in filter
    for ttype, value in stream:
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/lexer.py", line 192, in streamer
    for i, t, v in self.get_tokens_unprocessed(text):
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/lexers/python.py", line 386, in get_tokens_unprocessed
    pylexer.get_tokens_unprocessed(curcode)):
  File "/home/mainak/anaconda/lib/python2.7/site-packages/Pygments-2.0.2-py2.7.egg/pygments/lexer.py", line 785, in do_insertions
    for i, t, v in tokens:
  File "/home/mainak/anaconda/lib/python2.7/site-packages/pyface/ui/qt4/code_editor/pygments_highlighter.py", line 39, in get_tokens_unprocessed
    for item in action(self, m):
TypeError: 'NoneType' object is not callable
```

mainakjas, 2015-05-06:
I've trying updating sphinx and stuff like that but no luck :(

agramfort, 2015-05-06:
I get the same error as @mainakjas on my box

pyface/ui/qt4/code_editor/pygments_highlighter.py", line 39, in
get_tokens_unprocessed
    for item in action(self, m):
TypeError: 'NoneType' object is not callable

Eric89GXL, 2015-05-06:
Have either of you tried `git clone`ing https://github.com/enthought/pyface and doing a manually install of the latest version? might work

agramfort, 2015-05-08:
did anybody figure out what happens? it's a "bit" annoying not to be able
to build the doc while we're about to release...

Eric89GXL, 2015-05-08:
I can't replicate, and I don't think anyone has tried my proposed workaround yet (using cloned, built, installed, pyface)

Eric89GXL, 2015-05-08:
I'm not sure what we should do about the `nilearn` example not building, open an issue over there and hope they fix it soon...? Not sure I have time to dig into their code.

agramfort, 2015-05-08:
I opened an issue:

https://github.com/nilearn/nilearn/issues/577

Eric89GXL, 2015-05-08:
Yeah, it might also be an issue with `nibabel`, that's one thing I wasn't sure about.

agramfort, 2015-05-09:
PR sent https://github.com/nilearn/nilearn/pull/578

Eric89GXL, 2015-05-09:
Did you give someone else access to your GitHub account or something? It
says you actually made a commit and opened the PR yourself :)

Eric89GXL, 2015-05-14:
cloning pyface and installing the latest dev version fixed the doc building
bug on my anaconda system @mainakjas @agramfort

dengemann, 2015-05-14:
We should at some point clean up our dependencies a bit ...

Eric89GXL, 2015-05-14:
FYI this is only a bug for building doc, not using the package.

dengemann, 2015-05-14:
I know. But it's crazy that we rely on such hacks and high-level
dependencies to build docs. This is dangerous. If I could dictate I would
remove or isolate dependencies that go beyond the mne dependencies and the
regular scipy stack + testing + docs building.

Eric89GXL, 2015-05-14:
In this instance, pyface is a dependency of `sphinx` itself so there is nothing we can do. Did you have other dependencies you can think of?

dengemann, 2015-05-14:
Ah ok, got it. I think the last time we had some problems with pygraphviz,
but that only prevented some parts from building right? Take my statment as
future-oriented ;-)

agramfort, 2015-05-14:
ok works for me now with new pyface master...