mne-tools/mne-python #2065: Script/module for pipeline in the spirit of freesurfer's recon-all

lulopolar, 2015-05-06:
Hi Guys,
I wrote a little script to run a standard mne pipeline. The intention with this is to have a tool similar in spirit to the recon-all script in freesurfer. It also loads the relevant data when the files exist. Do you think this might be of interest to other people, or might be something that is worth adding to mne-python? The python file is here: https://gist.github.com/lulopolar/ad592cd35bb41055e94c

agramfort, 2015-05-07:
interesting. Can you document how to use it?

could become an mne python command in mne/commands but it should not be too
custom.

thanks heaps camilo for sharing !

lulopolar, 2015-05-07:
@agramfort I am glad this might be useful. The documentation is already in the function `pipeline`'s docstring, which I copied below. As for customization, at the moment it has probably many options. It would great to hear from the guys that follow the mne-python project to determine if some options are not needed.

    >>> import os.path as op
    >>> from mne.datasets import sample
    >>> from pipeline import pipeline
    >>> data_path = sample.data_path()
    >>> mne_rawdata_path = op.join(data_path, 'MEG/sample')
    >>> subjects_dir = op.join(data_path, 'subjects')
    >>> mne_working_path = op.expanduser('~')
    >>> subject = 'sample'
    >>> raw_fnames = 'sample_audvis_raw.fif'
    >>> bem_path = op.join(subjects_dir,
                           'sample/bem/sample-5120-5120-5120-bem-sol.fif')
    >>> trans_path = op.join(mne_rawdata_path, 'sample_audvis_raw-trans.fif')
    >>> save_prefix = dict(meeg='erase_sample_audvis', mri='erase_sample_mri')
    >>> overwrite = True
    >>> raw, events, epochs, evokeds, cov, src, fwd, inv, stc = pipeline(
            subject, raw_fnames, bem_path, trans_path, save_prefix=save_prefix,
            subjects_dir=subjects_dir, mne_rawdata_path=mne_rawdata_path,
            mne_working_path=mne_working_path, overwrite=overwrite
            )

    Now we have everything save to disk.  If we now want to just load the data
    and have fun with it, we can just do as follows.

    >>> mne_rawdata_path = mne_working_path
    >>> raw_fnames = 'samaudvis-filt-0.65-40.0-raw.fif'
    >>> overwrite = False
    >>> raw, events, epochs, evokeds, cov, src, fwd, inv, stc = pipeline(
            subject, raw_fnames, bem_path, trans_path, save_prefix=save_prefix,
            subjects_dir=subjects_dir, mne_rawdata_path=mne_rawdata_path,
            mne_working_path=mne_working_path, overwrite=overwrite
            )

agramfort, 2017-07-22:
gist disappeared. Closing

closed 2017-07-22 by agramfort