mne-tools/mne-python #2068: FIX: picks in GAT.predict()?

kingjr, 2015-05-07:
I just realized that the GAT object picks channels both at the ```fit``` and ```predict``` steps. 

In my opinion, we should at least keep the ```picks``` param as a ```gat.picks``` attribute after fitting and re-using it by default for the predicting case:
```
def predict(X, picks=None):
    if picks is None:
        picks = gat.picks
```

Another option would be to remove the channel picking from the GAT. This seems more like a preprocessing to be done independently of it, doesn't it?

@dengemann @cmoutard @mainakjas @MechCoder @jona-sassenhagen : any counter thoughts or specific use-case for which my intuition d be wrong?

Else, I'll deal with this FIX once https://github.com/mne-tools/mne-python/pull/1965 is done.

agramfort, 2015-05-07:
if you remove one param from the object you make me happy !

indeed picks can be selected before hand.

but we should remember the names of the channels used.

kingjr, 2015-05-07:
> indeed picks can be selected before hand. 

Ok 

> but we should remember the names of the channels used.

add ```gat.channels_``` ?

dengemann, 2015-05-07:
rather gat.ch_names_ or consistent with ICA .ch_names (but insoncsistent with sklearn).

kingjr, 2015-05-07:
I vote for the maximization of within-package consistency

dengemann, 2015-05-07:
> I vote for the maximization of within-package consistency

me too, because this is not relevant for the decoding aspects of the object. In ICA we have a similar split.

kingjr, 2015-05-21:
All fixed in https://github.com/mne-tools/mne-python/pull/2072 and https://github.com/mne-tools/mne-python/pull/1965.

closed 2015-05-21 by kingjr

dengemann, 2015-05-21:
you closed this @kingjr ?

dengemann, 2015-05-21:
ahh ok. I'm confused.